WEC-Sim/WEC-Sim #120: Multiple Condition Runs,

Sire79, 2016-08-11:
Hi WECSim team,
I am attempting to simulate a device with a multiple condition runs feature (wecSimMCR) with the wave data in an excel spreadsheet. 
My questions are as follows:

- Should the data in the spreadsheet be a matrix or vector (i.e. wave power matrix or wave heights and     wave time stamps)?
- How do I call the spreadsheet data to the model? 

Regards
Sire

kmruehl, 2016-08-16:
This is more or less the same question as #119 #121. 

For an example of how to set up and run wecSimMCR, refer to an example from the [exp branch located here](https://github.com/WEC-Sim/WEC-Sim/tree/exp/experimental_runs/1_freeDecay/heaveDecay/heaveDecay_linear_c450_cd0). The relevant files are:
wecSimInputFile.m
heaveDecayCase.mat

yuyihsiang, 2016-08-23:
See my answer for #119

kmruehl, 2016-08-31:
@Sire79  has your MCR question been resolved?

closed 2016-10-25 by kmruehl