WEC-Sim/WEC-Sim #121: Multiple Run simulation

SudharsanKalidoss, 2016-08-11:
How to write output in multiple run simulation? And How should input is to be given to wecsim through spreadsheet for multiple run simulation?

Sire79, 2016-08-12:
Hi Sudharsan,
If I understand you properly, output are given by the multiple run model that is run by running "wecSimMCR" input file. There is a command to call a spreadsheet to wecsim but it does not working properly for me.

Regards

yuyihsiang, 2016-08-24:
Hi SudharsanKalidoss

To write output in multiple run simulation, our suggest is to write matlab script in userDefinedFunctions.m file. Normally, we would use that to copy the output file in the output folder to another place and rename it. 

Thanks,

--Yi-Hsiang

SudharsanKalidoss, 2016-08-30:
Hi Yi-Hsiang,

    Can I write a script to copy and paste folder in userDefinedFunction.m? Or else is there any particular format to do the same?

Thanks,

Sudharsan

yuyihsiang, 2016-08-31:
I believe you can. Any MATLAB function will work in the userDefinedFunction.m file.

See MathWorks for more details: http://www.mathworks.com/help/matlab/ref/copyfile.html?requestedDomain=www.mathworks.com

--Yi-Hsiang

kmruehl, 2016-08-31:
@SudharsanKalidoss @Sire79 have your MCR questions been resolved?

closed 2016-10-25 by kmruehl