WEC-Sim/WEC-Sim #123: RM3 simulation convergence error

lianggege, 2016-08-16:
Hi everyone,
I am modifying the RM3 case to build a new numerical model. My modifications are as following;
1. replace the initial PTO by an extra  FIXED CONSTRAINT to make the two floaters rigid connected.
2. replace the initial FLOATING (3DOF) by a FLOATING (6DOF) CONSTRAINT to make to entire system free   
float on the water.
After the above modifications, I run the code and get very strange and interesting results.
The time history of 6DOF motions  seem all right at first, then the equilibrium position change rapidly. If the simulation time is long enough, then following error will be reported:

### Error using wecSim (line 232)
Derivative of state 'RM3.Floating_6DOF.CONSTRAINT_Bushing_Joint.Px.v' in block 'RM3/Floating
(6DOF)/CONSTRAINT (Bushing Joint)' at time 847.90000000000009 is not finite. The simulation will be
stopped. There may be a singularity in the solution.  If not, try reducing the step size (either by
reducing the fixed step size or by tightening the error tolerances)

And the two floaters seems not to be fixed together, as their motions are not exactly the same.
Does anyone have similar experience or could offer a help?
Thanks a lot!

Liang
[image]
[image]

kmruehl, 2016-08-16:
1. If the two bodies are indeed fixed, then why model them as two separate bodies instead of just one rigid body? This seems to be an unneeded complication in the model.
2. This is likely an issue with your time-step being too large. Try running some time-step convergence studies to pick a appropriate time-step, or run the simulation with a variable time-step (this is a new feature).

lianggege, 2016-08-17:
Thanks for reply!
I set the time-step 0.001s, which I believe is small enough. Unfortunately, the results remain unchanged.

yuyihsiang, 2016-08-23:
First, the response of each body is recorded at the CG for each body. Therefore, if the device is pitching, the heave and surge for the two body will be different even when the two bodies is locked. However, the pitch should be the same.

Regarding the instability, it will be helpful for us to investigate, if you can explain what is showing in the figure. HeaveSurge/Pitch?

Thanks,

--Yi-Hsiang

lianggege, 2016-08-25:
Hi, Yi-Hsiang, the figure above is surge motion. In fact, heave as well as pitch motions have the same problems.

yuyihsiang, 2016-08-26:
Hi Liang

This is caused by the quality of the hydrodynamic coefficients we provided here. In particular, the impulse response function never fully decays to zero for pitch for both bodies. In this case, it was particularly problematic for T=8~9 sec.

Typically, it would not matter if a viscous drag coefficient is included in your simulation or a PTO damping is given for the original example. 

My suggestions:
1. Specify a corresponding viscous drag coefficient for the two bodies in heave, surge and pitch.
2. If you are running regular wave, you can use "regular" instead of "regularCIC". 
3. If you are running irregular wave, you are most likely to be fine.

Let us know how it goes.

Thanks,

--Yi-Hsiang

kmruehl, 2016-11-03:
@lianggege has this been resolved?

closed 2016-11-15 by kmruehl