WEC-Sim/WEC-Sim #124: Hydrodynamic analysis of two floaters

lianggege, 2016-08-16:
Hi, everyone,
I am doing time domain simulation of the hydrodynamic interaction between a FPSO and a shuttle ship. Unlike wave energy system, such as RM3, the two floaters (FPSO and shuttle ship) in my study have only 
hydrodynamic interaction and no mechanical couplings exist. 
Could I ask whether my following modeling is correct and would anyone kindly offer a help if there is something wrong?
[image]

Thanks a lot
Liang

yuyihsiang, 2016-08-23:
Hi Lianggege 

Theoretically, it is correct. You can also try another option by connecting the Base of the Floating (6DOF)1 directly to Coon of the Global Reference Frame. 

However, it will be great if you can provide a little more information on what you are trying to investigate, such as the interaction between FPSO and shuttle ship or something else. There could be a simpler approach, depending on what you are trying to do here.

Thanks,

--Yi-Hsiang

lianggege, 2016-08-25:
Hi, Yi-Hsiang,
Thanks for your reply! 
I just want to calculate the motions of each floater when they are placed side by side.

yuyihsiang, 2016-08-26:
Hi lianggege

I see. It would be easier for me to give further suggestion if you are welling to sharing the model. I am more than happy to take a look if you can send me your model and input files.

Thanks,

--Yi-Hsiang

kmruehl, 2016-10-25:
@lianggege @yuyihsiang  has this issue be resolved?

closed 2016-11-03 by kmruehl