WEC-Sim/WEC-Sim #125: About running the model

rz311, 2016-08-25:
I got the error: The variant block 'RM3/Float/Hydrodynamic Body/B2B Subsystem/B2B' does not have same number of input ports
as Variant Subsystem block 'RM3/Float/Hydrodynamic Body/B2B Subsystem'.
Could anyone help?

yuyihsiang, 2016-08-26:
Which version of MTALAB are you using? Currently, the WEC-Sim model only supports MATLAB 2015b or later version.

rz311, 2016-08-26:
Hi,
It's 2013b. Do you think this problem is caused by the version problem?

Thanks

Steven

yuyihsiang, 2016-08-26:
ya, 2013b won't work. Many of our blocks developed in WEC-Sim lib use functions of SimMechanics (now SimScape Multibody in 2016a) that only exist in later version of MATLAB.

rz311, 2016-08-26:
Thanks.

That helps.

Steven

rz311, 2016-08-31:
Hi,
Again I got errors with a 2015B version: 
Caused by:
    Error using wecSim (line 232)
    Error in port widths or dimensions. Invalid dimension has been specified for input port 1 of 'RM3/Float/Hydrodynamic
    Body/Hydrostatic Restoring Force Calculation/Add'.
    Error using wecSim (line 232)
    Error in port widths or dimensions. Output port 1 of 'RM3/Float/Hydrodynamic Body/Hydrostatic Restoring Force
    Calculation/Displacement' is a one dimensional vector with 4 elements.

Thanks

Steven

kmruehl, 2016-08-31:
@rz311 are you running the RM3 tutorial (https://github.com/WEC-Sim/WEC-Sim/tree/master/tutorials/RM3), and have you added the wec-sim source code and all of the sub-directories to your path? If so, you should not have this error. 
-Kelley

rz311, 2016-08-31:
Hi,
I think so. Is this error coming from the mismatch of the dimension for different signals?

Steven

kmruehl, 2016-09-01:
The error is indeed a dimension mismatch error, but without seeing your model I cannot troubleshoot to identify the issue. Would you like to upload the model for the WEC-Sim team to review?

rz311, 2016-09-05:
Hi,
It’s just the tutorial I downloaded from the website directly…

Thanks

Stven

kmruehl, 2016-10-25:
@rz311 did this resolve your issue?

closed 2016-11-02 by j-vanrij