WEC-Sim/WEC-Sim #127: linear hydrostaticc,  heave pitch coefficients

krisRaj, 2016-09-22:
Hello,

Take for example non zero C35 (heave pitch coupling). This means C53=C35.  But I see C53 set as zero in the hydro_coeffs.linear_restoring_stiffness.   I also checked with [C]*[disp]  and forceRestoring and in these calculations too, C53 is zero.

Have I done in mistakes in importing the wamit data.

Thank you

Kumar

yuyihsiang, 2016-09-24:
Hi Kumar

In the wamit .out file there is no C53. Therefore, when we read the .out file, we have to set C53=C35. If you look at the rm3.h5 file in the tutorial, C53=C35 (note indexing there start from 0). So I am surprise that you see C53 set as zero in the hydro_coeffs.linear_restoring_stiffness. I'll check some WEC-Sim code later this weekend to make sure there is nothing going on and get beck to early next week.

Thanks,

--Yi-Hsiang

krisRaj, 2016-09-24:
Thank you, Yi-Hsiang for the reply
Since the read is through bemio, and we users leave the .out file from WAMIT untouched, I will wait for your answer. 
Kumar

yuyihsiang, 2016-09-25:
Hi Kumar

I took a look on the WEC-Sim and BEMIO models. The C53 should equal to C35 and was defined in BEMIO. All the test cases I have got shows C53=C35 as well as C54=C45. It will be good to check the version of WEC-Sim and BEMIO that you are using. Are you using the latest version that has BEMIO in MATLAB?

Thanks,

--Yi-Hsiang

krisRaj, 2016-09-26:
Hi Yi-Hsiang,
Sorry I did not post my findings earlier. The issue is only happening in the older version of Bemio. (bemio that i downloaded in 10/23/2015) . For the later Bemio, that I downloaded in early this year (march/2016? ) the reading is correct, meaning C35 = C53.  Thanks again for your reply (on a sunday!)

Kumar

yuyihsiang, 2016-09-26:
No worries. Good to know the code is fine. In the older version, I think we forced C53=C35 in the WEC-Sim code. But later we decided to do that inside BEMIO so that it can be consistent between WAMIT, Nemoh and AQWA. 

Also want to let you know that we have recently (in August) released a new version of BEMIO, which is now in MATLAB. You can also use that in the future. 

Cheers,

--Yi-Hsiang

closed 2016-09-26 by yuyihsiang