WEC-Sim/WEC-Sim #128: How to include Morison element

krisRaj, 2016-09-28:
Hello,

1)Tried to include morison element in  wecsim latest (and the one just older) like this:
body(1).morrisonElement.ca={0 0 0};
body(1).morrisonElement.cd={0 0 8};
body(1).morrisonElement.characteristicArea=3;
body(1).morrisonElement.VME=0.0;
body(1).morrisonElement.rgME=0.0;

but the forceMorrisonAndViscous field is empty.

2) Is there any way to have more than one element in a body (say one for surge and one for heave).

Thanks

Kumar

yuyihsiang, 2016-09-29:
Hi Kumar

Sorry for the late reply. we are really busy this week since it is the last week of FY16. Given that I haven't use Morison Element for a while, I'll take a look quickly on Monday to answer your question.

Thanks,

--Yi-Hsiang

krisRaj, 2016-09-29:
No worries , Yi-Hsiang. 
Thank you. Kumar

krisRaj, 2016-10-11:
Hi Yi-Hsiang,

I think I know the answer to 2nd question of my post. 

Can you give me some pointers as to why the following did not work in the latest version (and in the one before)
 body(1).morrisonElement.ca={0 0 0};
body(1).morrisonElement.cd={0 0 8};
body(1).morrisonElement.characteristicArea=3;
body(1).morrisonElement.VME=0.0;
body(1).morrisonElement.rgME=0.0;

Thank you
Kumar

yuyihsiang, 2016-10-11:
Hi Kumar

I believe the information you put in did not follow the required code stricture. I copy pasted the definition of the parameters from the bodyClass.m file (Ln41~46). 

morrisonElement   = struct(...                                          % Structure defining the Morrison Elements
                                   'cd',                 [0 0 0], ...               % Viscous (quadratic) drag cd, vector length 3
                                   'ca',                 [0 0 0], ...               % Added mass coefficent for Morrison Element          (format [Ca_x Ca_y Ca_z], default = [0 0 0])
                                   'characteristicArea', [0 0 0], ...               % Characteristic area for Morrison Elements calculations (format [Area_x Area_y Area_z], default = [0 0 0])
                                   'VME',                 0     , ...               % Characteristic volume for Morrison Element (default = 0)
                                   'rgME',               [0 0 0])                   % Vector from center of gravity to point of application for Morrison Element (format [X Y Z], default = [0 0 0]).

As you can see, both the characteristicArea and rgME need to be a vector not a scalar. Let me know if you have more questions.

Thanks, 

--Yi-Gsiang

krisRaj, 2016-10-12:
Hi Yi-Hsiang,
I am still having trouble. So I do this:
body(1).morrisonElement.cd(3)=8;
body(1).morrisonElement.characteristicArea(3)=16.0;

and I query in command window:
>> body(1).morrisonElement

ans = 

                    cd: [0 0 8]
                    ca: [0 0 0]
    characteristicArea: [0 0 16]
                   VME: 0
                  rgME: [0 0 0]

So the reading of input is good.

But the output.bodies(1).forceMorrisonAndViscous(:,3) , field is zero (actually  all fields are zeros)

Thanks 
Kumar

bradling, 2016-10-12:
Did you make sure to set `simu.morrisonElement = 1;`? It's default value is `0`, and needs to be set to true to calculate the morison drag terms.

yuyihsiang, 2016-10-12:
HI Kumar

As Brad said, you will need to set simu.morrisonElement = 1, and I have confirmed that if you did not specify that and the default value is 0, which will make everything equal to 0 for the morrison element term.

--Yi-Hsiang

krisRaj, 2016-10-12:
Thank you, Yi-Hsiang and  Brad,

It works now.

Kumar

closed 2016-10-28 by yuyihsiang