WEC-Sim/WEC-Sim #129: Problems displaying/interpreting geometry and/or AQWA data

gui-paredes, 2016-09-30:
Hi
I am trying to model a simple vertical cylinder oscillating in waves with a 3DOF constraint in WEC-SIM 2.1, using MATLAB 2015b.  The properties of the cylinder are as follows: radius of 2.2 m; height of 15 m; draft of 10 m; centre of gravity 6 m below the water level. I created the geometry in ANSYS DESIGN MODELER and obtained the hydrodynamic coefficients from AQWA. I converted the aqwa ah1 and lis files to h5 using the matlab script for BEMIO, with most options set to default. I exported the geometry to .stp from design modeller and then converted to .stl using FreeCAD.

Everything seemed to work well, but after the calculations, the geometry of the cylinder is shown as fully submerged (2-3 m below the water level). Is this a bug, or modelling issue? And what can I do to try to fix it or find out where the problem comes from? Is this only a visualisation problem, or does it also affect the computations?

I can send the full model data if required.

P. S. There was also an issue reading the stl file. WEC-SIM interpreted the initial geometry as having a radius of 2200 m and a height of 1500 m, and I had to scale down the geometry to a radius of 2.2 mm and a height of 15 mm for WEC-Sim display it correctly.

gui-paredes, 2016-09-30:
I think I found (partially) the issue. The geometric representation of the bodies in WEC-SIM is done by adding the coordinates of the centre of gravity to the coordinates in the mesh file. If the geometry is already defined in the equilibrium position, with submerged parts having negative z coordinates and above water parts having positive ones, the final representation will not be the proper one in WEC-SIM. Additionally, as I understood it, the "equilibrium" condition for the mass does not perform a proper equilibrium check; it only computes the mass of the submerged volume read from the hydrodynamic data. I suggest that you add a condition to check the position of the geometry in WEC-SIM, determine the submerged volume and displaced mass, then compare it with the attributed mass in wecSimInputFile. Please correct me if I am wrong and provide me some feedback on this.

yuyihsiang, 2016-09-30:
Hi

Yes, you are correct. The required stl file in WEC-Sim simulation also assumes an origin at the center of gravity for each body. We will evaluate this again and see if we want to add other options.

When the mass="equilibrium", it is also correct that we compute the mass from the submerged volume read from the hydrodynamic data if the linear hydro model is used. However, if the nonlinear hydro option is used, we then calculate it based on the stl submerged volume. 

Thanks,

--Yi-Hsiang

kmruehl, 2016-11-03:
Reiterating @yuyihsiang's comment, `body(1).mass = 'equilibrium';` solves for the mass based on the displaced volume of the *.STL file, and therefore requires the *.STL file to have a good mesh. `body(1).mass = value;` sets the mass equal to a value, and does not solve for the displaced volume. Before running any simulation, we recommend running a hydrostatic test with no waves to verify that the model is in hydrostatic equilibrium. 

Your understanding of the coordinate systems is also correct. 

Please let us know if you have any further questions.

kmruehl, 2016-11-15:
@gui-paredes does this answer your question?

gui-paredes, 2016-11-15:
Thank you for your replies @kmruehl and @yuyihsiang , which did answer my questions. :)

NMartinez9, 2018-03-25:
Hi everyone,

I have a similar problem simulating a bottom hinged flap. I am using Nemoh and the Mesh rutine that take account only the submerged part of the body, and the origin of coordinates at z=0. I also use the FreeCAD Salomé for the geometry .STL file, but placeing the origin of coordinates at the flap's CG.
The simulation results are not as i spected so like to ask if I need to do some more changes on the Nemoh files and how I can do so.  

Thank you

closed 2018-04-03 by kmruehl