WEC-Sim/WEC-Sim #130: [Question about viscous drag]

Alfian-TD, 2016-10-08:
Hi, 

The function body(i).viscDrag.cd and body(i).viscDrag.characteristicArea require input in vector length 6. How to put the coefficient into vector length 6? What does every column in vector length 6 represent? Let's say we have drag coefficient of 2.5, how do we put it into vector length 6? 

Is there a way to calculate a drag coefficient and a characteristic area in WEC-Sim? 

Warm regards, 
Alfian

Alfian-TD, 2016-10-13:
Hi Yi-Hsiang, thanks for the answer. I have one more question. I was wondering about the characteristic area as well. What is characteristic area? How do you define it? Is it the surface area of the whole body? If so, does this mean the vector of 6 for the characteristicArea consist of 6 same numbers for each degree of freedom? 

--Alfian

yuyihsiang, 2016-10-13:
Hi Alfian

Generally, the first three characteristic areas can be the device projected area in X, Y, Z direction. The three rotational ones are the tough ones, to make the unit consistent, it should be the corresponding projected area  multiplies R^3, where R is a characteristic length. The bottom line is that it really does not matter, as long as you have a Cd is defined using the same assumption. The paper below give an example of how you can estimate the drag coefficients and define the characteristic areas.

Cozijn H., Uittenbogaard R., and Brake E. ter, 2005, “Heave, Roll and Pitch Damping of a Deepwater CALM Buoy with a Skirt,” 15th International Offshore and Polar Engineering Conference, ISOPE, Seoul, Korea, pp. 388–395.

Thanks,

--Yi-Hsiang

stevenshine, 2016-11-03:
hi ＠Alfian-TD 
i have same questions with u 
you how to solve them questions ,can i discuss with u ？

kmruehl, 2016-11-03:
@stevenshine @Alfian-TD I recommend looking into related fields to better understand the definition of drag coefficient and characteristic area. Definition of drag is a common issues for any field with fluid flow. There are some seminal publications for drag in oscillatory flow, such as waves, but definition of drag can be tricky. The characteristic area is typically defined as "the cross-sectional area of the body perpendicular to the flow direction.", and drag coefficients have been defined for various shapes in the literature. 

For more information on the theory/application of viscous drag in WEC-Sim, please refer to the following:
- [Theory](http://wec-sim.github.io/WEC-Sim/theory.html?highlight=viscous%20drag#morison-elements)
- [Application](http://wec-sim.github.io/WEC-Sim/code_structure.html?highlight=drag#bodyclass)

Alfian-TD, 2016-11-06:
Hi Yi-Hsiang, 
Thank you very much for your explanations and reference. That's very helpful for me. I've been learning from the paper you've recommended, yet still not fully understand about it. I've also been learning to make use of SimFlow (open-source) in order to get drag coefficients, and now is still in progress of learning. It might take quite a long time for me to learn, but in case it's faster than expected I'll tell my progress. Once again, thank you very much for the help. 

Hi Steven, 
Yeah, sure. What're we going to discuss? 

Hi KM Ruehl, 
Thank you very much for your explanations. I've been referring to what you've recommended, and it's very helpful. Thanks. 

--Alfian-TD

closed 2016-11-15 by kmruehl