WEC-Sim/WEC-Sim #132: Morison element;   Irregular waves

krisRaj, 2016-10-24:
Hi,
I am getting errors similar to what Brad got  in this post .
https://github.com/WEC-Sim/WEC-Sim/pull/103  which he fixed.  (https://github.com/WEC-Sim/WEC-Sim/pull/103/commits/369f79e48e9804d9034d5983776f7101c9696871)

Since I do not want to upgrade my wecsim version to the latest can anybody share with me this WECSim_Lib.slx that was uploaded march 16/2016, 

Thanks very much.

Kumar

krisRaj, 2016-10-25:
Hi Yi-Hsiang,

I hope I can bug you one more time.  So finally I got the latest version of wecSim and added the morison element and mooring matrix, but I am still having issues with irregular waves: I get this error,
////////////////
Error using wecSim (line 232)
Derivative of state 'h_and_f_rev2.Floating.CONSTRAINT_Planar_Joint.Px.v' in block
'h_and_f_rev2/Floating/CONSTRAINT (Planar Joint)' at time 0.0 is not finite. The
simulation will be stopped. There may be a singularity in the solution.  If not, try
reducing the step size (either by reducing the fixed step size or by tightening the
error tolerances)
///////////////

Last time this happened, you had modified the WECSim_lib.slx ( https://github.com/WEC-Sim/WEC-Sim/issues/84) .  (I don't think it is the time step. this is very small).  I wonder if you can help me with the mods. this time too, or let me know how this is done in the .slx file. I will try to do this at my end  

Thank you very much.
Kumar

krisRaj, 2016-10-28:
Looks like this issue is resolved. Hacked the .out file to remove some high frequencies. runs nicely (so far).

Cheers

Kumar

yuyihsiang, 2016-10-28:
Hi Kumar

Are you running linear or nonlinear hydro? The fix for WECSim_lib.slx ( #84) is for nonlinear hydro option. However, WAMIT should have taken care of it if linear hydrodynamics option is used. I am surprised to see it still give you errors. It will be great if you can send me you run, including the hydrodata, the slx model and other files. I'll take a look into this.

Thanks,

--Yi-Hsiang

yuyihsiang, 2016-10-31:
Hi Kumar

I believe I have found the issue. It was in the Morrison Element model, which also implement the wave theory the way we did in issue #84. I'll try to fix it later since it was written in in simulink format not in MATLAB code and will take a little more time. For now the code itself should be fine, and we just have to avoid very small wave frequency data.

In addition, I also noticed that you used mooring matrix, but did not specify a connection location. If you don't  specify a connection location, the default value is 0,0,0, which I don't think it is correct for your case. I would suggest you add a line in wecSimInputFile.m file 

mooring(1).ref = [a location in x, y z];

This will give you the correct mooring matrix force at a give location. Please note that in order to have it correctly you will need to use the latest WEC-Sim, which I updated last week.

Thanks,

--Yi-Hsiang

krisRaj, 2016-11-01:
Thanks very much Yi-Hsiang, for the heads up on the mooring(1).ref.  With the new block and mooring.ref. I get exactly the same surge as the previous wecsim (good version , but no mooring block)

Kumar

kmruehl, 2016-11-03:
@krisRaj has this issue been resolved? 

For future reference, if you do not want to download the latest version of WEC-Sim, but there is one particular file you want to download, you can find that particular file in the master repository and manually download it. For example, to get the master version of the WEC-Sim Simulink library file, follow this link and click 'download':
https://github.com/WEC-Sim/WEC-Sim/blob/master/source/lib/WEC-Sim/WECSim_Lib.slx

closed 2016-11-15 by kmruehl