WEC-Sim/WEC-Sim #134: Mooring matrix reference issue

bradling, 2016-10-26:
Hello,

I am having an issue with the mooring matrix block. I am defining my mooring in my input file as:
`mooring(1) = mooringClass('mooring1');`
`mooring(1).matrix.k(3,3) = 1e6;`
`mooring(1).ref = [0 0 -30];`

I get the following heave response when simulating in no waves:

[image]

I would think there would be no heave response in this case, as there is no mooring pretension, and the body mass is set to equilibrium. Have I properly configured the mooring object?

Thanks,

yuyihsiang, 2016-10-28:
Hi Brad

Just update the mooring matrix block. It should be fixed now. I just need to subtract the position for the mooring by the mooring.ref, which somehow was missed in the version.

Thanks,

-Yi-Hsiang

bradling, 2016-11-01:
Hi Yi-Hsiang,

Looks like this fixed it. Thanks,

closed 2016-11-01 by bradling