WEC-Sim/WEC-Sim #136: PTO force control

gui-paredes, 2016-11-01:
Can anyone please help me in the application of user defined pto forces in WEC-SIM using the "Translation PTO actuation force" block? A simple example would help me a lot! :)

kmruehl, 2016-11-03:
We have examples of using the force actuation block on the [WEC-Sim_Applications repository](https://github.com/WEC-Sim/WEC-Sim_Applications). The examples are have are related to application of PTO-Sim. This feature is also [documented on our website here](http://wec-sim.github.io/WEC-Sim/features.html#pto-sim).

gui-paredes, 2016-11-04:
I already figured how to input user defined forces and change the behaviour of the buoy, but I got another problem: the simulation results indicate that there was no power extracted (output.ptos.forceInternalMechanics and output.ptos.powerInternalMechanics are both zero matrices). I am using the simple simulink model shown below
[image]
 
I am not using any ptoSimInputFile.m file and in wecSimInputFile the value of pto(1).c  is set to zero.

Am I doing something wrong?

kmruehl, 2016-11-15:
@gui-paredes the implementation of the `Translational PTO Actual Force` block is correct, however if you want to see the externally applied force,  `output.ptos.forceInternalMechanics` and `output.ptos.powerInternalMechanics` are the not the relevant outputs. If you look under the mask of the `Translational PTO Actual Force` block (below), you'll see that the `Internal Mechanics` subsystem calculates the force and power based on internal mechanics only. In this case, the internal mechanices are pto(1).c and pto(1).k, so if they are set to zero, the outputs for internal force and power will also be zero. 

[image]

The relevant output is `output.ptos.forceActuation` for force, and you would have to calculate the power yourself by multiplying the actuation force times the velocity. Hope this helps!

gui-paredes, 2016-11-15:
Thank you for your reply @kmruehl .  I  used a work-around to get the power and force to be saved to the variables powerInternalMechanics and forceInternalMechanics using the file userDefinedFunctions. Maybe not ideal, but will save computing time when using MCR.

closed 2016-11-30 by kmruehl