WEC-Sim/WEC-Sim #138: MATLAB - BEMIO functions - Radiation_IRF

sabible, 2016-11-02:
When running the Matlab BEMIO functions on either the RM3 or OSWEC tutorials, the function Radiation_IRF results in NaN for all entries of hydro.ra_K.  The WAMIT import of RM3.out or OSWEC.out appear successful and I don't see spurious results in any of the other saved variables.

j-vanrij, 2016-11-02:
Yes, we have several MATLAB BEMIO test cases, they are located within the WEC-Sim directory:
‘…\WEC-Sim\tutorials\BEMIO\...’

I have seen this result before, it occurs when w_max is specified in the IRF function, Radiation_IRF(hydro,t_end,n_t,n_w,w_min,**w_max**), as the maximum frequency from the BEM data. If you give [] for w_max instead, it will use the same maximum value, but will not give you the NaN’s. This is a quirk that should be fixed in the future, but my guess is that if you just use [] instead, this will solve your problem…

sabible, 2016-11-02:
Jennifer, indeed that was it, genius.  Thnx.

closed 2016-11-02 by sabible