WEC-Sim/WEC-Sim #140: Rhino-Nemoh-WecSim

MattSharman, 2016-11-04:
Hi All, 

Just wondering if anyone has had any success importing a mesh from Rhino exported as a .GDF. And converting this to a mesh Nemoh can understand?

All advice, best practice and tips welcome!
Thanks, 
Matt

j-vanrij, 2016-11-04:
Hi Matt,

The python based BEMIO functions, http://wec-sim.github.io/bemio/, have mesh conversion capabilities that will do this for you, and this is currently how I convert a .gdf mesh into a NEMOH mesh. You would have to download and install the python based BEMIO, but once you do, there are tutorials available and the mesh conversion is quite easy. (We would like to transfer these mesh conversion capabilities from the python based BEMIO to the new MATLAB based BEMIO, but have not had a chance to do so yet…).

Alternatively, the NEMOH site (http://lheea.ec-nantes.fr/doku.php/emo/nemoh/start) list several additional options – GDFMesh.m and MESHMAGICK. I have not tried these, and so cannot vouch for them, but they could also be a simple solution.

Thanks,
Jennifer

bjrose86, 2016-11-04:
> doku.php/emo/nemoh/start) list several additional options – GDFMesh.m and

Hi Matt,

I've had success using GDFMesh.m referenced above.

If I recall, there are couple subtleties. NEMOH assumes symmetry about the
x-z plane, so the GDF file should only define the left or right half of the
body. Also, make sure the coordinates are consistent with the NEMOH's wave
direction convention:

http://lheea.ec-nantes.fr/lib/exe/fetch.php/emo/nemoh/general_notations_and_conventions.pdf

Brian

MattSharman, 2016-11-07:
Thanks Jennifer and Brian!

I'd like to stay within Matlab if i can help it so I'm giving GDFMesh a go. 

I've created a GDF mesh in Rhino, in line with the subtleties you mentioned Brian. But I'm getting all sorts of file reading errors and I'm struggling to work out whether it's my file which is the issue or GDFMesh code (E.g. issues like compatibility with Rhino GDF meshes, version updates in Matlab etc.). 

Has anyone got a working .GDF file that works with GDFMesh i can use to help debug? And did you have to change much within GDFMesh.m to get it to work Brian?

Many Thanks, 
Matt

bjrose86, 2016-11-07:
Matt,

Looking back at my code and comparing it to the original, it looks like I
had to make a few small edits to the "Reading and Grouping" section in
order to parse the GDF file correctly.

Brian

sabible, 2016-11-08:
Matt,
I've borrowed and modified a python script I found in the NEMOH forum, attributed to Julius Schay, to export directly from Rhino to NEMOH format(s).  There are actually 2 scripts, one exports to the format used by the "Mesh.exe" Nemoh routine (which you need to run in order to calculate some hydro coefficients used in Bemio), while the other exports to the format used by the "preProcessor.exe" and later routines.  The "Mesh.exe" routine actually creates an additional mesh as well, but I have found the Nemoh created meshes to be inconsistent and I know what I am getting when I see the mesh in Rhino so I like using these instead.  They are attached.  Hope you find this method helpful.

[rhinoMesh.zip](https://github.com/WEC-Sim/WEC-Sim/files/579271/rhinoMesh.zip)

bjrose86, 2016-11-08:
Matt,

Here's the GDFMesh.m code I modified. Hope this helps. 

Brian

[GDFMesh.zip](https://github.com/WEC-Sim/WEC-Sim/files/579394/GDFMesh.zip)

MattSharman, 2016-11-09:
Hi All,

Many thanks for your help!

Finally gotten it to work. Two changes required, the first was indeed in the reading and grouping section, similar to your modifications brian. The other issue I had was with the GDF (created in Rhino) was creating Nemoh meshes that were inside out, with the panel normals pointing inside the body. This was due to the GDF panel vertices being defined in the wrong order. Reversing this order fixed this. 

Thanks Sabible for that script, I'll give it a go at some point, does it save you much time?

Thanks again all!
Matt

sabible, 2016-11-09:
Matt, yes, it is the most efficient method I have used.  In time I believe WEC-SIM will evolve to include geometry import + integrated meshing + Nemoh calcs, but the time line for that is uncertain.  In the meantime, Rhino exported mesh straight thought to Nemoh calcs, and then back to WEC-Sim is where we stand.  It is really painless to store that script and run it from within Nemoh.  Access is from the "tools" dropdown.  

Another complication as pointed out by others (#129) is that Nemoh assumes/requires the geometry origin (or at least z = 0) to be at the mean sea level, whereas WEC-SIM currently assumes the origin is at the COG, which requires you to export a modified STL geometry for use in WEC-SIM.  But I think that too will change when developers integrate everything into WEC-SIM, or possibly before.  

Stewart

kmruehl, 2016-11-15:
@MattSharman does this answer your question?

MattSharman, 2016-11-15:
Yes, Thank you all! Will be back with more I'm sure!

closed 2016-11-15 by MattSharman