WEC-Sim/WEC-Sim #141: Controlling stroke length

SudharsanKalidoss, 2016-11-08:
Hi,

    I'm trying to control relative motion between float and spar in RM3 model. For example, I want the float to oscillate +1.5 m or -1.5 m relative to spar plate. Is there a way to do that in WecSim. 

Thankyou

kmruehl, 2016-11-15:
@SudharsanKalidoss WEC-Sim does not currently have the ability to define endstops between relative bodies. I think the best way to handle this would be to use the `Translational PTO Actuation Motion` block, and set up a loop that senses if the relative motion exceeds your endstops, and force the position to be equal to the max value. 

@yuyihsiang @j-vanrij Can you think of another solution?

bradling, 2016-11-23:
Hi,

 I have had some luck modeling end stops as very stiff springs. It doesn't prevent the motion from exceeding the translation limits, but if the stiffness is set high enough the limits will only be exceeded by a small amount. I use the PTO actuation force block, and define my own simulink submodel that does:

if position > max position
force = -k*(position - max position)
else
force = 0

Then I still use the 'pto.c' parameter to set the pto damping, and can still access output.ptos(1).powerInternalMechanics. The "end stop" force will not change this power value.

I hope this is helpful!

kmruehl, 2016-11-30:
thanks @bradling that's another good suggestion.

closed 2016-11-30 by kmruehl