WEC-Sim/WEC-Sim #142: Any plans to incorporate Nemoh into WEC-Sim as was tried in OpenWARP?

sabible, 2016-11-08:
Just curious, as it would enable more efficient optimization trials through the use of Matlab inherent optimization tools.

kmruehl, 2016-11-15:
@sabible we currently parse NEMOH results into the wec-sim code using the BEMIO processor. Can you elaborate on what you mean by further incorporating NEMOH into WEC-Sim?

sabible, 2016-11-15:
Sure, I could envision integration happening in two ways:  
- developers could port the NEMOH fortran code to Matlab language and execute within WEC-Sim similar to what was done when BEMIO went from Python to Matlab routines.  User simply imports geometry file and does not need to externally execute Nemoh. 
- or, Matlab/WEC-Sim could be used to externally launch the existing Nemoh fortran code and return the results to the Matlab working directory.  Still requires user to have local copy of Nemoh code and file directory stuff.  Maybe WEC-Sim could help in the creation of the input files/directories, to make sure both simulations are compatible and making the same assumptions.
Either way, the advantage would be that existing Matlab functionality for optimization could be used to drive the iterative CAD creation - CAD Export - NEMOH - BEMIO - WEC-Sim process.  

Having Matlab drive CAD creation and export can be easy or tricky depending upon the users CAD program.  What would be ideal is if there is a open source parametric CAD creation tool that plays nice with Matlab.  

Beyond the impact on driving optimization, I believe having Nemoh integrated natively within WEC-Sim can increase usership, as many WEC developers can't pay for Wamit et al., and have a barrier to usage with having to deal with the non-GUI nature of Nemoh.  And in general, the less number of softwares I have to manually work with the more likely I am to use WEC-Sim.  
Integration would also improve the accuracy of user simulations, as the way it is setup right now I think there are many places for users to make a mistakes in combining use of Nemoh and WEC-Sim, including differing assumptions regarding geometry orientations, unclear direction regarding what Nemoh options to run to be compatible with WEC-Sim assumptions (DOF, # of frequencies, etc), etc..  And without experimental data for individuals to verify their WEC-Sim model against, it is shooting blind whether they have set everything up correctly.  In my experience the most successful commercial softwares do everything they can to make simulation as accessible and as full proof as possible.  These are just suggestions and not at all criticism of the great work done so far on WEC-Sim.

kmruehl, 2016-12-07:
@sabible Thank you very much for your feedback. Better integration of WEC-Sim with a BEM solver is desirable, and would certainly making setting up/running a WEC-Sim simulation more seamless. At this point, there is no ongoing effort to do this, but it is certainly a functionality that would benefit the WEC-Sim user community at large. 

@lawsonro3 @yuyihsiang @j-vanrij do you have anything you would like to add? This is something we should make note of in future code development discussions.

j-vanrij, 2017-01-26:
Thanks so much for the suggestions, and I apologize for the long overdue response.

The short answer is - It is a great idea, but probably not something that we have the time and resources to commit to. However, as an open source code, if this is something you really want to see happen, you, or anyone else, are welcome to make the changes required to do this!

The longer answer is - Although the BEM->BEMIO->WEC-Sim workflow appears unnecessarily complicated, inefficient, and error prone, there are reasons that the process is the way it is, which would make integrating each of the steps difficult, although certainly not impossible, as outlined below. 

1. We would like to remain compatible with Nemoh, WAMIT, and AQWA, as we know we have users that utilize each of these codes, rather than focus exclusively on Nemoh.
2. Typically, a BEM simulation (and BEMIO) is run once, and utilized by multiple WEC-Sim simulations. So, an integrated, once through, process of Nemoh-BEMIO-WEC-Sim would not be efficient for most WEC-Sim runs. Of course, flags could be created, if there are already Nemoh and BEMIO output files in existence, and so unnecessary to rerun, but then we are essentially returning to the current process.
3. It is necessary to check the BEM output (and often rerun it) either through BEMIO or by self-inspection, to verify that it is of good quality, before running it in WEC-Sim, which creates another somewhat mandatory break in the workflow.
4. Although, I for one think that the MATLAB-Nemoh run scripts are awesome, creating a CAD geometry from MATLAB is a real challenge (as you said), and with the current  MATLAB-Nemoh wrapper codes, really only the most basic geometries can be created. This, again, makes it almost mandatory to keep this as a separate step.

closed 2017-01-26 by j-vanrij