WEC-Sim/WEC-Sim #144: KEEP GETTING THESE ERRORS WHEN I RUN THE SIMULATION ON RM3

ChaitanyaGadodia, 2016-11-23:
1)The variant control 'sv_b1_hydroBody' of the variant block 'RM3/Float' must be a condition expression or name of a Simulink.Variant object in the global scope workspace of the model.
2)The variant control 'sv_b1_nonHydroBody' of the variant block 'RM3/Float' must be a condition expression or name of a Simulink.Variant object in the global scope workspace of the model.

ThommyBee, 2016-11-28:
Hi ChaitanyaGadodia,

are your body blocks in the RM3 simulink file still library linked?
Sometimes, the simulink model is saved while the library link was releaved/broken to adjust the body block. If then the variant indicating which type of body you want to model is in the workspace of matlab, the simulink model will be saved with a the variant control of the body block being defined in the model rather than in the wecSimInput File. The next time you run the model, you typically get this error.

Try to replace the body block in the simulink model by doing the following:

0) Clear all variables in your workspace or simply restart Matlab
1) Open the Simulink model named "RM3.slx". The block at the very top called "Float - body(1)" should be a wec-Sim library linked block with variants enabled. Thus, non of the small icons on the bottom left corner in this particular block should be shown as broken or missing. 
2) Select the block and delete it.
3) Open the library browser (Ctrl + Shift + L) and navigate to the wecSim library.
4) Chose the Rigid Body block and drag it into your system. connect it to the lose wire from before.
5) Run the model

Hope that helps
Thomas

kmruehl, 2016-11-29:
@ThommyBee thanks for your suggestion, refreshing the WEC-Sim Library and making sure the Similink library blocks are linked, and have not been overridden is a good starting point. 

@ChaitanyaGadodia what version of matlab are you using, and what version of wec-sim are you using? To check the matlab version, [follow these steps ](http://wec-sim.github.io/WEC-Sim/getting_started.html#matlab-toolbox-requirements)

kmruehl, 2016-12-05:
@ChaitanyaGadodia does this resolve your issue?

ChaitanyaGadodia, 2016-12-06:
@kmruehl yes, the method told by @ThommyBee worked!
@ThommyBee thanks a lot.

kmruehl, 2016-12-06:
@ChaitanyaGadodia wonderful. I'm glad the suggestion helped. Making sure the library block are linked and up-to-date is always a good starting point.

closed 2016-12-06 by kmruehl