WEC-Sim/WEC-Sim #145: Potential MCR Speed Improvement - Changing Hydro Data?

bradling, 2016-12-01:
Hello WEC-Sim Community,

This is more of a question about a potential improvement than an issue. I often use wecSimMCR to sweep through various regular wave cases. I have noticed that a significant portion of the run time in regular waves is dedicated to loading the h5 file for each instance (~20 seconds of the 30 second total run time per case). It would be possible to load the .h5 file only once when using wecSimMCR, and that 20 seconds of run time could be eliminated for each additional case tested.

So my question is this:

Does anyone use wecSimMCR to change the hydrodynamic file being used?

If that is not a common use case, then it would be easy to modify wecSimMCR to load the h5 file once. Another option would be to include a flag in the simu object that tells wecSimMCR whether or not it needs to reread the h5 file each time.

If this is something the WEC Sim team is interested in including in the code I would be happy to make a pull request including this improvement. However, if this is something the development team is not interested in including, I don't want to waste my time making a pull request.

kmruehl, 2016-12-01:
@bradling I think this is a great suggestion, and as you mentioned, it would reduce the run time for MCR runs a lot. If this is a change you are interesting in making, I recommend you make a pull request to do so. I would also suggest that you make it a simu flag so that it is an option that users can toggle on/off,  with the default such that the h5 file is only loaded once. This would allow flexibility as there may be instances when a user would want to load it between runs (the only example I can think of is if the user has different BEM runs they want to compare). Also, you'll need to check what variables are being cleared between MCR runs, and verify that the hydrodynamics properties are stored between runs (but nothing else from the previous run).

bradling, 2016-12-01:
OK I will work a little bit on this when I have time, and then make a pull request, keeping in mind your suggestions when implementing this.

kmruehl, 2016-12-05:
@bradling thank you

Toby-D, 2016-12-15:
@bradling: Good call on not reloading the .h5 data. 

It's also worth noting that most (~90%) of the CPU time associated with loading the .h5 data is actually spent checking the MATLAB version. 

Lines 59-62 of h5load.m address a compatibility issue with versions from 2008 and earlier. Using MATLAB's profile tool on a representative run, it looks like the current software version gets checked 379 times while loading the .h5 data, which adds up to 8.6 s of the 9.6 s it took to load the data. 

So if you're not planning on going back to R2008 anytime soon, you can replace those lines with:
%   v = ver('MATLAB');
%   if datenum(v.Date)>datenum('04-Aug-2008')
  objtype = objtype+1;
%   end

_Be sure to leave the third line active_.

(A more robust approach, for anyone who is seriously committed to compatibility, would be to load the  matlab version as an optional argument in h5load and bypass the version check if "v" already exists...)

Cheers,
Toby.

bradling, 2016-12-16:
Toby, interesting observation. I could be wrong here, but I think that WEC-Sim requires matlab 2015b and later to run properly anyways. Perhaps that fact could be used as an argument to eliminate these version checking lines and improve the runtime of h5load? 

A third option that I have been thinking about would be to move the hydrodynamic data from an h5 file to a .mat file. This might make sense now that bemio is being transitioned into matlab. Maybe BEMIO could be configured to write either h5 files or mat files? And then WEC-Sim could be written to read either, calling different functions based on the file extension. Using mat files in matlab is the fastest way to save and load data, since it is mathwork's native file format.

@kmruehl your thoughts?

kmruehl, 2016-12-16:
@Toby-D Thanks for the info, that's a good catch. h5load.m is an open source script not written by the WEC-Sim team, so we never reviewed it for efficiency (or removed legacy version checks). I removed those lines and have [pushed to master ](https://github.com/WEC-Sim/WEC-Sim/commit/f5249d52c6b3e5fd9a548f584d0e63dd53f0c7d7)as we do not support matlab version prior to 2015b, but expect the user to [perform this check themselves](http://wec-sim.github.io/WEC-Sim/getting_started.html#matlab-toolbox-requirements).

@bradling omitting the *.h5 files altogether, and using the native matlab *.mat data structure is an option we discussed internally with the Lab wec-sim team, and we chose to stick with the *.h5 file format. I honestly cannot recall why this decision was made. @j-vanrij can you shed some light on this?
@Toby-D Thanks for the info, that's a good catch. h5load.m is an open source script not written by the WEC-Sim team, so we never reviewed it for efficiency (or removed legacy version checks). I removed those lines and have  pushed to master as we do not support matlab version prior to 2015b, but expect the user to [perform this check themselves](http://wec-sim.github.io/WEC-Sim/getting_started.html#matlab-toolbox-requirements).

@bradling omitting the *.h5 files altogether, and using the native matlab *.mat data structure is an option we discussed internally with the Lab wec-sim team, and we chose to stick with the *.h5 file format. I honestly cannot recall why this decision was made. @j-vanrij can you shed some light on this?

j-vanrij, 2016-12-16:
There are several reasons we decided, at the time, to not just using a .mat data structure. 1) When we first moved the BEMIO functions from the python based code to MATLAB based functions, we want to allow users to use either. 2) There are still functions in the python code that have not been converted to MATLAB yet (mesh conversion and visualization). 3) And, some users prefer the HDFViewer to check the hydrodynamic data. 
I think, eventually, soon, it will make sense to do away with the .h5 data structure and use a .mat data structure but I can’t say when that will be.

kmruehl, 2016-12-17:
@j-vanrij thanks!

closed 2017-01-25 by kmruehl