WEC-Sim/WEC-Sim #147: wave class JONSWAP

stevenshine, 2016-12-07:
JONSWAP Spectrum
how to set wave typle  JONSWAP spectrum
i read imformation  on wec-sim web but  i dont very clear understand ''This is the WEC-Sim default spectrum'' mean
[image]

do u help me?
ths u

kmruehl, 2016-12-08:
Yes, that's the correct way to specify a JS spectrum. I just reran the JS spectrum for the RM3 tutorial, and it worked for me. Here is a what I specified in the WEC-Sim input file:

```
waves = waveClass('irregular');       %Create the Wave Variable and Specify Type
waves.H = 2.5;                        %Significant Wave Height [m]
waves.T = 8;                          %Peak Period [s]
waves.spectrumType = 'JS';
```

As for the online documentation, I apologize for the comment about a default spectrum. That was an outdated note (which I just removed), WEC-Sim no longer has a default spectrum, the spectrum type must be specified in the waveClass by defining the `waves.spectrumType` variable.

stevenshine, 2017-11-20:
@kmruehl 
Hi
I am trying wec-sim1.2 about JONSWAP Spectrum
but it get error 
where i should modify?
thx u
[image]

kmruehl, 2017-11-29:
@stevenshine please download or clone the [latest version of WEC-Sim master](https://github.com/WEC-Sim/WEC-Sim ). We no longer support v1.2 of WEC-Sim as the [current version of WEC-Sim is v2.2](https://github.com/WEC-Sim/WEC-Sim/releases). As a matter of fact, we plan to release a new version this week, so please keep your eyes peeled. 
Thanks!

stevenshine, 2017-11-30:
@kmruehl 
 thank u 
i try WEC-Sim 22 version still have same promble
and then i set 180 and 225 degree 
[image]

[image]

kmruehl, 2017-11-30:
@stevenshine It looks like your issue is not with the JONSWAP spectrum, but actually with the BEM solution. The `*.h5` file used for your WEC-Sim run must have the desired wave directions (`waves.waveDir` = 180 and 225) included. 

Also, looking back at the previous error, it appears that your `*.h5` file may not have the impulse response function (IRF) included. In order to run any irregular wave case the `*.h5` file must include the IRF. 

Please rerun your BEM solution to include the desired wave directions and then run BEMIO to include the IRF calculation. The BEMIO documentation is [available here](http://wec-sim.github.io/WEC-Sim/advanced_features.html#bemio).

stevenshine, 2017-12-01:
@kmruehl 
hi
 I set waveDir 180 and 225 degree and simulink in the same time at WECSIM2.2 but still get error

[image]
how to modify ?  
thanks u

kmruehl, 2017-12-01:
@stevenshine Did you rerun NEMOH to include those wave directions prior to running BEMIO? 

The BEMIO error says that the IRF was not written. Can you ZIP your model with the NEMOH output, the BEMIO file, and the WEC-Sim model. We can try to debug it that way. Thanks.

stevenshine, 2017-12-02:
yes ,when i run NEMH also include those wave directions
The WECSIM and NEMOH files store on the address
My case located on RM3 file on wecSIM Folder
it is Google Drive 
https://drive.google.com/drive/folders/1EClltjunw9iTmCBJK-sZJFwPh2Zn4lRd?usp=sharing
thanks u  again

kmruehl, 2017-12-12:
@stevenshine Thank you for attaching your model files. It appears as though you have two separate issues:

1. BEMIO is unable to run for your RM3 NEMOH results. Were you able to debug the issues you previously had with running the NEMOH simulation? If so, @j-vanrij or @dforbush2 will help troubleshoot the BEMIO parser for NEMOH. However, if the issue is related to troubleshooting the NEMOH run itself, you'll need to [contact the developers of NEMOH](https://lheea.ec-nantes.fr/logiciels-et-brevets/nemoh-toolbox-192933.kjsp). 

2. Based on the screenshots you attached previously, the `wecSimInputFile.m` was not properly defined. You must specify `waves.waveDir = <#>;`  as demonstrated in the attached input file for the [RM3.zip](https://github.com/WEC-Sim/WEC-Sim/files/1553590/RM3.zip). Once you've run the 180 and 225  degree cases in NEMOH and generated the `*.h5` file using BEMIO, you will be able to specify `waves.waveDir = 180 ;` or  `waves.waveDir = 225;` in the input file. However, you cannot specify it as `waveDir = [2 180 225];` as shown in the above screenshot. Also, it appears as though you uploaded the v2.2 release of WEC-Sim to google drive, and not the WEC-Sim model run of the RM3 you have been referencing. In the future, please *.ZIP the specific run, not the source code. This will help us troubleshoot the issue. We've just [released a new version of WEC-Sim](https://github.com/WEC-Sim/WEC-Sim/releases), we recommend using it moving forward.

stevenshine, 2017-12-13:
If I use same hydrodynamic coefficient in WECSIM and then run regular and irregualr cases
the regular can simulate in WECSIM but irregular cant run in WECSIM.
When u open my files on Google Drive, Did u find any error/bug about Nemoh?
I don’t knoe what there are problems in Nemoh.
thanks u again

kmruehl, 2017-12-13:
@stevenshine, below are my comments. @j-vanrij is going to take over from here regarding any BEMIO questions. 

1. Your `.h5` file does not include the impulse response functions. This is why WEC-Sim will run for regular waves, and not for irregular waves. This is likely due to the NEMOH error you posted previously (shown below)
[image]
Also, your `*.h5` files only includes data for wave directions equal to 0, 180 degrees. If you want to run WEC-Sim with 225 degrees, you need to include this direction in your NEMOH run. 

2. There are [NEMOH results for the RM3 along with the BEMIO run available here](https://github.com/WEC-Sim/WEC-Sim/tree/master/tutorials/BEMIO/NEMOH/RM3). You should be able to modify these files to meet your needs and @j-vanrij will answer any questions regarding the BEMIO run for NEMOH.

j-vanrij, 2017-12-13:
Hello @stevenshine,

I took a look at your NEMOH, BEMIO, and WEC-Sim files and found two issues:

1. In your Nemoh.cal file please change ‘! Fluid density’ to ‘! Fluid specific volume’ and rerun bemio. It is not necessary to rerun Nemoh, the bemio Read_NEMOH function is just looking for specific phases to read in, so if they are changed, it will not find them and will stop before calculating the IRF’s (as it did in your case). Once you have your new .h5 file, with the IRFs, move it to your WEC-Sim hydroData folder.
2. In your wecSimInputFile.m please change ‘wave.waveDir’ to ‘waves.waveDir’ 

And with those changes your WEC-Sim model should run – at least it did for me.

Thanks!

stevenshine, 2017-12-14:
HIi @j-vanrij 
Acctually i dont really understand ur mean
i try to change Fluid desity to Fluid specific volume but still get error messages
[image]
how to modify it 
Thanks u again

j-vanrij, 2017-12-14:
@stevenshine 

OK, the Nemoh and WEC-Sim files are attached.
[NEMOH.zip](https://github.com/WEC-Sim/WEC-Sim/files/1557985/NEMOH.zip)
[RM3.zip](https://github.com/WEC-Sim/WEC-Sim/files/1557987/RM3.zip)

The files that were modified, as stated above, are Nemoh.cal, bemio.m, NEMOH.h5, and wecSimInputFile.m. Using these inputs, the simulation appears to be working...

Thanks.

stevenshine, 2017-12-14:
HI@j-vanrij 
Thanks u help me modify these files
now the file can run irregular now,but i still  want to ask u some questions.
i want to  simulate the WECSIM  run two degree waves.waveDir at one case
How about set two degrees at one case in WECSIM?
Thank u again

stevenshine, 2017-12-14:
hi @j-vanrij 
thanks u help me modify theses files
the files can simulate now. but i still have some questions.
1.how to set two degrees waveDir at same case and run WECSIM
Thanks u again

j-vanrij, 2017-12-14:
@stevenshine, you are very welcome. 

For further WEC-Sim questions, please refer to the WEC-Sim documentation and previously resolved issues to see if your question has already been asked and answered. For example, the following previous issues answer your question about implementing wave directions in WEC-Sim.

[https://github.com/WEC-Sim/WEC-Sim/issues/12](url)
[https://github.com/WEC-Sim/WEC-Sim/issues/202](url)

However, to summarize, WEC-Sim can only simulate one wave direction at a time, and there must be a BEM solution for the wave direction of interest, as specified in the WEC-Sim input file with waves.waveDir.

Thank you.

closed 2017-12-14 by j-vanrij

stevenshine, 2017-12-19:
@kmruehl 
sorr, i also have some questions
i can modify the jonswap wave's gamma ?
how about set gamma in wecsim 
thanks u

kmruehl, 2017-12-20:
How to specify gamma in the `wecSimInputFile `is described on the wec-sim website [here](http://wec-sim.github.io/WEC-Sim/code_structure.html#irregular).

stevenshine, 2017-12-25:
@kmruehl  @j-vanrij 
ok thaks u
i still have questions,
when i try to storage JS wave  paraview ,but WECSim get error.
[image]
 help！ thanks u again

stevenshine, 2017-12-25:
@kmruehl @j-vanrij 
thanks u . i can run that smoothly