WEC-Sim/WEC-Sim #148: Real-Time Support / Porting

awmackay, 2016-12-14:
I'm currently working on a Wave Energy Scotland project and we're considering using WEC-Sim as part of a system model.

I wondered if WEC-Sim would be compatible with Simulink Real-Time?  I've not yet got access to the required software to check myself and hoped one of the developers / contributors could help.

yuyihsiang, 2016-12-15:
Hi Alan

This is an very interesting question. My direct reaction is that you should be able to do so as long as the WEC-Sim simulation is faster than real time, which is true unless you are using nonlinear hydrodynamics option. But I think this is a question to MATLAB. I have send a MATLAB support we know your question and will let you know if I heard anything from him.

Thanks,

--Yi-Hsiang

kmruehl, 2016-12-15:
Reiterating @yuyihsiang's response: I do not see any reason why Simulink Real-Time would not be compatible with WEC-Sim, but to the best of my knowledge no one has tried it yet. I'm curious to hear what MathWorks responds.

closed 2017-02-06 by kmruehl