WEC-Sim/WEC-Sim #151: Distributed Diffraction / Radiation Pressure

joaocruz-ca, 2017-01-25:
Hi there

We have been developing a few items for WEC-Sim and we were wondering if there are plans to upgrade the distributed pressure outputs to include also the diffraction and the radiated pressure contributions (the latter making use of the body dynamics as a post-processing step).

Our understanding is that at present only the hydrostatic and the incident (FK) components are distributed, although all the main components are considered in the WEC dynamics. Most flow solvers offer the diffraction and radiation potentials as an output, thus the extension to the current capability seems possible. 

Keen not to replicate any ongoing development hence the request!

Thanks for the feedback 

Joao

yuyihsiang, 2017-01-26:
Hi Joao

We actually have done some of that to certain extend using Nemoh, but not in WAMIT. There was a version of Python based of BEMIO (we can try to find it), and it has the option to read in pressure (but not velocity potential) from Nemoh and we have subroutines to calculate the time-history pressure distribution based on the WEC-sim response. We had a paper at [METS 2016](https://github.com/WEC-Sim/WEC-Sim/files/732941/Michelen.et.al.-.2016.pdf) that demonstrated how we use the pressure distribution to calculate the structure loading on the device. 

However, we did not include that in the MATLAB version of BEMIO and we did not continue the BEM-to-FEA process because it becomes more complicated when multiple bodies and multiple PTOS joints are involved for the FEA side. Nevertheless, it will be great to know what's your objective and we could possibly find common ground to jointly work on that. If you are interested, we can also try to find the python version of BEMIO that has the read option for pressure from Nemoh.
 
Thanks,

--Yi-Hsiang

joaocruz-ca, 2017-01-27:
Hi @yuyihsiang 

Thank you for the feedback and for the link to the METS paper - if you have the Python script that would be useful. 

We have started a MATLAB equivalent but along the lines of your comments we are conscious of the issues of adding multiple bodies. Our background experience is in making the link between WAMIT, MOSES and AQWA data - but we were now moving towards NEMOH (hence the request).

We are mostly interested (short-term) in single hydrodynamic body problems - so the Phyton script would be useful (if we convert it to MATLAB we would of course let you know). The re-meshing / mapping between the FEA package and the WEC-Sim output that follows could also be an idea for joint work.

Thanks again

Joao

yuyihsiang, 2017-02-08:
Hi Joao

We finally find the code that read in the pressure from Nemoh. It is not in the released BEMIO. we had a separate version and you can down load it from

https://dl.dropboxusercontent.com/u/12367608/ECM-master.zip

- All the functions should be in the single file 'bem2fea.pyx' (Line: 359).
- There are example scripts that use these functions in the 'bem2fea/' and ' distPressure/' folders. They load Nemoh results, load motion time history data, and calculate distributed pressure time histories.

Hope this helps

Thanks,

--Yi-Hsiang

joaocruz-ca, 2017-02-09:
Hi @yuyihsiang 

Thank you for the link - I will download and check it, if we come up with a generic script I will let you know.

Best Regards

Joao

closed 2017-02-09 by kmruehl