WEC-Sim/WEC-Sim #152: Excitation Force Phase

joaocruz-ca, 2017-01-25:
Hi there

We were re-running some basic verification steps, namely the RM3 WEC-Sim model using the provided WAMIT data and an equivalent model with nemoh data.

Using the same wave heading, we noticed that the angular response and exciting moment time series are 180deg out of phase (between the WAMIT and nemoh related models). The phasing issue does not emerge in the translation modes.

We were wondering if this has been experienced by other users? Changing the phase associated with the diffraction problem in nemoh ('Compute_Wave' routine) allows the phase alignment to occur, but if this a regular issue perhaps there is potential to adjust it directly in WEC-Sim?

Thanks for the feedback

Joao

kmruehl, 2017-01-25:
@joaocruz-ca which versions of BEMIO and WEC-Sim are you using? Also, are you using your own NEMOH data for the RM3 model, or are you using the [WEC-Sim provided RM3 NEMOH](https://github.com/WEC-Sim/WEC-Sim/tree/master/tutorials/BEMIO/NEMOH/RM3) run? The WEC-Sim provided data should be correct because this sign convention was resolved in the [MATLAB version of BEMIO](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions/BEMIO). We believe that the legacy python version of BEMIO (which we are no longer supporting) may still have the phase error.

joaocruz-ca, 2017-01-26:
Hi @kmruehl - and thanks for the speedy response. 

Using the MATLAB version of BEMIO (and our NEMOH data) the inconsistency is solved. This does avoid us changing the convention in NEMOH in the future, which is good! 

We will discontinue the python version that we had previously modified.

Thanks again,

Joao

kmruehl, 2017-01-26:
Good. I am glad this resolved your issue.

closed 2017-01-26 by kmruehl