WEC-Sim/WEC-Sim #155: Morison Equation Coefficients

MattSharman, 2017-02-01:
Hi All,

I'm modelling a point absorber WEC concept and am looking to better define the viscous drag on the device. 

Does anyone have any advice on generating the added mass coefficient and drag coefficients for simple shape buoys, preferably without the requirement of further CFD?  What coefficients/methods do people use? And can you calculate the added mass coefficients from the BEM results?

Any advice would be greatly appreciated!
Matt

kmruehl, 2017-02-06:
@MattSharman the added mass coefficient will come from your BEM solution. In WEC-Sim, this is parsed by the [MATLAB BEMIO pre-processor](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions/BEMIO).

The drag coefficient is user-defined, and typically determined from experiments or CFD. However, there are some publications that specify drag coefficients. Refer to previous issues #139 #130

MattSharman, 2017-02-09:
Right ok, does WEC-sim take the added mass coeffs from the .h5 file itself or does this need to be done manually and entered into the wecSimInputFile under body(1).morrisonElement.ca? 

Thanks for pointing me in the right direction!

kmruehl, 2017-02-09:
I just realized that I may have misunderstood your question. There are several different ways in WEC-Sim to apply added mass and damping/drag:

1. [Linear added/mass damping based on BEM solution](http://wec-sim.github.io/WEC-Sim/theory.html?highlight=drag#sinusoidal-steady-state-response-scenario) - This is done automatically by WEC-Sim in the [bodyClass](https://github.com/WEC-Sim/WEC-Sim/blob/master/source/objects/bodyClass.m) within the 'readH5File method. This method reads the hf5 file written by BEMIO, and directly imports the added mass and dampingfrom the hf5 file.

2. [Additional linear damping/non-linear drag](http://wec-sim.github.io/WEC-Sim/theory.html?highlight=drag#linear-quadratic-damping-forces) - This can be applied on each body by defining `body(1).linearDamping` in each DOF for addition linear damping, and/or `body(1).viscDrag` for each DOF for non-linear drag

3. [Morrison Elements](http://wec-sim.github.io/WEC-Sim/theory.html?highlight=drag#morison-elements) - Morrison Element drag/added mass can be applied on each body by defining `body(1).morrisonElement`. For this, `body(1).morrisonElement.ca` `body(1).morrisonElement.cd` `body(1).morrisonElement.characteristicArea` `body(1).morrisonElement.VME` and `body(1).morrisonElement.rgME` must be defined. 
Where:
`.VME` refers to the characteristic volume for Morrison Element (default = 0)
`.rgME` refers to the vector from center of gravity to point of application for Morrison Element (format [X Y Z], default = [0 0 0]).

The best way to understand the differences between these options is to see where these coefficients are applied within the Simulink body block (shown below). If you dig around in the Simulink model you'll see where/how these coefficients are applied in the model. 

[image]

kmruehl, 2017-02-22:
@MattSharman does this answer your question?

MattSharman, 2017-03-02:
Yes thank you, much appreciated!

closed 2017-03-02 by MattSharman