WEC-Sim/WEC-Sim #156: Force  relation

stevenshine, 2017-02-06:
Using wecsim softwave, set H=0.5  T=125s
there are wired condition
when forceExcitation is small and then the mooring force is biger
this is normal condition?
or this is out of calculate range ? 
both of force are relate to  equations ,right?
ths u 
[image]
[image]

kmruehl, 2017-02-06:
In order to help understand these results, I'll need to know some more information about your model setup and BEM solution. 

1. What is the frequency range of your BEM solution, and what does the solution look like around T=125s? A wave period of 125s is very long and possibly out of the stable range of your BEM solution. 
2. How is the mooring force defined? If it's a stiffness, then the mooing force will be dependent on the WEC displacement.

kmruehl, 2017-02-22:
@stevenshine does this answer your question?

stevenshine, 2017-02-23:
@kmruehl 
Maybe, this is out of calculate range. 
the BEM calculate by NEMOH.
BTW 
what do  effect of one-Order Hydrodynamics and effect of Second-Order Hydrodynamics  influences
?
are there about second-Order Hydrodynamics Program language source?
ths u

kmruehl, 2017-02-23:
I agree, my guess is that the the wave period of 125s is outside of the range of your stable BEM solution. 

The non-linear affects are detailed [on the wec-sim website here](http://wec-sim.github.io/WEC-Sim/features.html#non-linear-hydrodynamics). They include non-linear hydrostatic and non-linear Froude-Krylov forces based on the wetted surface of the STL mesh. Implementation can be found in the bodyClass and within the Simulink body block.

closed 2017-02-23 by kmruehl