WEC-Sim/WEC-Sim #158: Porting to plain Matlab: What does B2B block do exactly?

crobarcro, 2017-02-09:
Hello,

I'm porting WEC-Sim to use only plain matlab for the calculation of forces, and using MBDyn for the multibody dynamics. You can see the current status of this work [here](https://sourceforge.net/p/rnfoundry/hgrepo/ci/default/tree/wave/matlab-octave/wec-sim/). I'm essentially extending the bodyClass to also include methods for calculation of the dynamic forces. I copied the bodyClass to a new class hydrobody which has the extra methods. I've also had to tweak the simulationClass for this which isn't uploaded anywhere yet.

I am working my way through making sure the force results match WEC-Sim (I'm using the RM3 example as a reference at this stage) but am having some trouble with the added mass forces, and am hoping you could clarify a couple of things for me. 

In my Matlab code, to calculate the added mass I have the following:

`F_addedmass = body.hydroForce.fAddedMass * accel(:);`

I haven't yet applied the equivalent of the transport delay.

where `accel` is a (6 x n) matrix (when doing body to body interactions), and therefore `accel(:)` is a (6*n x 1) column vector, which appears to be the right format. The issue is that I cannot get the right forces with this. When I compare the forces ffrom simulink (before post-processing) the forces for the first body are correct (match the original WEC-Sim output) but for body 2 are very slightly different. 

If i then do the post-processing like:

```
for iBod = 1:simu.numWecBodies
    body(iBod).restoreMassMatrix
    output.bodies(iBod).forceTotal = output.bodies(iBod).forceTotal + output.bodies(iBod).forceAddedMass;
    output.bodies(iBod).forceAddedMass = body(iBod).forceAddedMass(output.bodies(iBod).acceleration,simu.b2b);
    output.bodies(iBod).forceTotal = output.bodies(iBod).forceTotal - output.bodies(iBod).forceAddedMass;
end; clear iBod
```
The forces are then all wrong (scaled up too much). Is there anything obvious I might be missing here?

My other question is, what is the purpose of the transport delay on the accelerations for the added mass calc?

kmruehl, 2017-02-09:
`simu.b2b` turns on/off [body2body interactions](http://wec-sim.github.io/WEC-Sim/features.html#body-to-body-interactions). If body2body interactions are on, the dimension of the velocity and acceleration terms change, and thus the matrix definition of added mass and damping changes as well.  Refer to the above website link for more info on b2b.

The lines you referenced from `wecSim.m` are re-assigning the time-history of the added mass force and the summation of the total forces (_if_ body2body interactions are on) in order to account for the cross-coupling affects. To understand why this is necessary requires digging around in the implementation of the force summation in the Simulink body block. The reason for the delay is that the added mass implementation in Simulink creates an algebraic loop, and the workaround for this is to add a delay. 

@yuyihsiang do you have anything you want to add?

crobarcro, 2017-02-10:
Actually, I've managed to check everything and am now getting identical output, so thanks again for the help, but my question is answered now.

kmruehl, 2017-02-13:
Yes, you are correct that the b2b SIMULINK block's purpose is to auto-generate the correct signal connections in order to arrange the velocity/acceleration in the correct structure. The dynamic response takes the form of [x,y,z,rx,ry,rz] for each body, and when b2b is turned on, with N total hydrodynamic bodies, the data structure has 6*N rows - the same as the data structure you posted. 

Glad you were able to get good agreement.

10472008, 2017-05-11:
Hi everyone,
I still cannot understand how B2B works.
The figure below is my test↓

[image]
[image]

the comparison of Forces

[image]

The Radiation damping and added-mass force of Body2 do not match, although Body1 seems fine.
Thank you for your helping.
Kind Regards.
Steve

crobarcro, 2017-05-11:
This confused me too, but If you look at wecSim.m you will see that post-processing is done on the added mass forces to determine the 'correct' values after simulation completes. No doubt one of the devs will explain this better.

10472008, 2017-05-11:
Hi,Crobarcro

The 'Test' is built by me, and Hydro coefficient is the same as WEC-Sim but the results of Body2 is different between both.
My guess is that my 'Coupled-Radiation Force' might be wrong, so I was wondering how does B2B works,
and cannot find any formulation about B2B.
Thanks your response.
Steve

kmruehl, 2017-05-11:
Have you verified that you are looking at the correct velocity/force signals? What B2B does is change the matrix sizes, and as a result the signals of interest are different. When B2B is off, they are defined as a [6x6] matrix, but if two bodies are interacting it will be a [6x12] matrix. For more info, refer to the [B2B documentation here](http://wec-sim.github.io/WEC-Sim/features.html#body-to-body-interactions).

B2B changes the size of the matrices, and combine the body(1) and body(2) velocity and  body(1) and body(2) acceleration signals. For example, the combined velocity signal is vel[6x12] is the velocity of body(1) is vel1[6x6], and the velocity of body(2) is vel2[6x6]. So in the combined matrix, the heave velocity of body(2) is vel(3,9). The same is true of the corresponding radiation force, etc. 

Here's a screenshot of how the singals are combined in the B2B variant subsystem when `simu.b2b=1`.
[image]

10472008, 2017-05-12:
Excuse me, I still cannot figure it out.
This is my equations which are refer to the B2B of WEC-Sim to calculate the radiation damping force.

[image]

Then, I implement Simulink model to excuse these equations. The result is much worse than first time.
May the mistake happened in equations or?
[image]

Thank you for your helping.
I really appreciate it.
Best Regards.
Steve

kmruehl, 2017-05-12:
First, my sincere apologies, I inverted the matrices. I _meant_ that they are [12x6] (not [6x12]). Also, the B2B WEC-Sim uses is based on the Cummins' impulse response function formulation, not a constant coefficient multiplication, so that is one difference. Using the WEC-Sim notation, B2B radiation force corresponds to: 

Radiation force on body(1)
Fr_3 = Fr_33+Fr_39, where Fr_33 = v_33 * fr_33 and Fr_39 = v_99 * fr_39

Radiation force on body(2)
Fr_9= Fr_99+Fr_99, where Fr_99 = v_99 * fr_99 and Fr_93 = v_33 * fr_93

Where * refers to convolution, Fr is the time-domain radiation force, fr is the radiation impulse response function, and fr_39=fr_93. 

The same is true for the expansion of the added mass, there is also a coupling term for it that takes the same form, but uses multiplication (not convolution).

10472008, 2017-05-15:
It doesn't matter. On the contrary, I appreciate your helping.
I go into the simulink of WEC-Sim to confirm what calculation does it use when I set B2B=1 and waveClass=regular (not CIC),and this is what I saw.

[image]
[image]

Maybe I'm wrong,but it seems like coefficient calculation method is taken.
The reference provided by you is very useful,although like many references I saw before. 
The reason that I don't tend to use convolution method is because It take too many time and not more accurate than coefficient calculation.  
Thank you for your helping.
Steve

kmruehl, 2017-05-17:
@10472008 I did not realize you were not using CIC, in that case, you are correct the B2B is determined using direct multiplication based on the radiation and added mass coefficients of the incident wave period, `waves.T`

Radiation force on body(1)
Fr_3 = Fr_33+Fr_39, where Fr_33 = v_33 * fr_33 and Fr_39 = v_99 * fr_39

Radiation force on body(2)
Fr_9= Fr_99+Fr_99, where Fr_99 = v_99 * fr_99 and Fr_93 = v_33 * fr_93

Where * refers to _multiplication_, Fr is the time-domain radiation force, fr is the radiation impulse response function, and fr_39=fr_93.

10472008, 2017-05-18:
Ms. Ruehl:
Thank you for your helping and contribution.
Kind Regards.
Lee

kmruehl, 2017-05-18:
@10472008 does this resolve the discrepancy?

closed 2017-05-18 by kmruehl