WEC-Sim/WEC-Sim #159: Morison Element state derivative 0 at time = 0.0

bradling, 2017-02-10:
Hello,

I am getting an error when trying to use morison drag terms in an irregular wave simulation. If I turn off morison drag then the error does not occur. The error message is:

`Error using wecSim (line 248)
Error occurred while simulating the model 'single_float_model' in rapid
accelerator mode. Rerun the simulation in normal mode to diagnose this
error. Continuous state derivative of 'element 3' is non-finite`

I spent some time tracing signals in the morison drag calculation block. I believe it to be the same error as was happening in https://github.com/WEC-Sim/WEC-Sim/issues/84, but its occurring in the Morison drag orbital velocity calculation. I am under a bit of a time crunch to make get this working, so I am going to rewrite the orbital velocity function in matlab code instead of simulink, and use the deep-water formula for large wavenumbers.

bradling, 2017-02-22:
See pull request https://github.com/WEC-Sim/WEC-Sim/pull/164. This seems to fix the problem.

I do have an additional question though. Why is the irregular wave amplitude (`waves.A`) multiplied by a factor of 2 when input to the orbital_velocity calculation:

[image]

This same factor does not exist in other calculations using `waves.A`, for example in the `waveClass` method `waveElevIrreg`, we have:

`tmp=sqrt(obj.A.*df);`
`tmp1 = tmp.*real(exp(sqrt(-1).*(obj.w.*t + obj.phaseRand)));`
`obj.waveAmpTime(i,1) = t;`
`obj.waveAmpTime(i,2) = sum(tmp1);`, 

and in the Nonlinear F-K calculation we have (where `AH = waves.A`):

`z=(center(:,3)-elv).*wDepth./(wDepth+elv);`
`f_tmp = rho.*g.*sqrt(AH(i)*dw).*cos(k(i).*center(:,1)-w(i)*t-phaseRand(i));`
`f = f + f_tmp.*(cosh(k(i).*(z+wDepth))./cosh(k(i).*wDepth));`

Is this difference due to how the orbital velocity calcuation implemented, or is it an error?

bradling, 2017-03-06:
@yuyihsiang  Any update on this? I see you approved the pull request that changes the calculation of orbital velocity to use deepwater wave approximation. But what about the term `2*A` discussed above?

yuyihsiang, 2017-03-07:
Hi Brad

The 2*A is simply just how the spectrum was defined. As shown in the Theory section in the document, the equation for Fext includes a factor of 2.

-Yi-Hsiang

bradling, 2017-03-07:
I am confused then. Why isn't that factor of 2 found in the water surface elevation calculation? Or in the linear excitation force calculation? Or the Nonlinear FK force calculation? They can't both be correct, right?

bradling, 2017-03-08:
To be clear, I am discussing irregular wave specifically. This is the case where `waves.A` corresponds to the wave spectra.

yuyihsiang, 2017-03-09:
I see that now. It is wired I agree. I'll check the block tomorrow when I get to the office.

yuyihsiang, 2017-03-09:
Hi Brad

I believe it is a bug and thank you for finding that. It should not have the factor of 2 there. the factor of 2 was included when calculating waves.A. I am updating the code.  

Thanks,

--Yi-Hsiang

bradling, 2017-03-09:
Great, thanks for looking into it further. For the cases I had been working with, the change in results was negligible.

One last thing before I close this issue, I was looking through the simulink model and when I updated the orbital_velocity calculation it appears I connected a few outputs of the 'orbital_velocity' block to the incorrect blocks. See image below. I can make a pull request tonight fixing this.

[image]

closed 2017-04-25 by kmruehl