WEC-Sim/WEC-Sim #161: Why adjustMassMatrix?

Toby-D, 2017-02-10:
Hi WEC-Sim Team,

Before the model is run in Simulink, the body mass and added mass are adjusted in BodyClass\adjustMassMatrix. And the forces, masses, and added mass values are correspondingly corrected in the post-processing. What's the rationale for this?
(This maybe relates to part of Issue #158 .)

I ask because I'm simulating a de-ballasting sequence, in which I pump water out of a submerged tank. During this sequence, the ballast body moves such that its added mass changes significantly, in addition its mass. So I need to:

1. Adjust the mass of the body during the simulation.
2. Adjust the added mass of the body during the simulation.

My thought was to accomplish (1) by using a [mass actuator](https://www.mathworks.com/help/physmod/sm/mech/ref/variablemassinertiaactuator.html). I then realized that I would need to separately control body.hydroForce.storage.mass in the Restoring Force subsystem. (Related question: Why does WEC-Sim apply weight manually instead of turning on gravity in the Global Reference Frame?) And I planned to implement the changing added mass in the Wave Radiation Forces block. But it's complicated by the fact that a body's mass and added mass are mushed together. 

I'm sure there's a good reason for the present formulation. (Maybe something to do with numerical stability?) It would be helpful to know what the reasoning was before I go changing things around.

Thanks!

Toby.

kmruehl, 2017-02-13:
@Toby-D I'll try my best to tackle all of these Qs. @yuyihsiang feel free to add more... 

Q1. Why do we adjust the added mass? 
Simscape Multibody does not allow definition of different mass values in each translational DOF. For the moment of inertia, this is possible [Ixx, Iyy, Izz], but mass cannot be defined in 3 DOF as [mx, my, mz]. The way we overcame this issue was by adjusting the added mass at each time-step using the `adjustMassMatrix` method in the `bodyClass`. We add the diagonal terms of added moment of inertia to the rotational DOF moment of inertia `obj.momOfInertia = obj.momOfInertia+tmp.fadm(4:6)';`. However, for the diagonal added mass terms, we combine the added mass in each DOF into one term `tmp.adjmass = sum(tmp.fadm(1:3))*adjMassWeightFun;,` and add the combined added mass to the total body mass `obj.mass = obj.mass + tmp.adjmass;`.  This is not the most elegant solution, but this is the most stable solution the WEC-Sim Lab team has found. If you have suggestions on a way to improve this, we are interested in hearing them!

Q2. Why does WEC-Sim apply weight manually instead of turning on gravity in the Global Reference Frame?
When WEC-Sim v1.0 was released there was not an option to turn gravity on automatically, so we did it manually and never changed the formulation in future WEC-Sim releases

Q3. Mass actuator?
I have never used this block before, but it may work. However you will have to be careful that however it is implemented does not negatively impact Q1

Toby-D, 2017-02-13:
Thanks @kmruehl! That's really helpful. I didn't realize, for example, that adjustMassMatrix was acting at each time step. If I come up with anything that seems better, I'll let you know, but it sounds like your time has thought it through pretty well already.

kmruehl, 2017-02-13:
@Toby-D I just realized I misspoke, adjustMassMatrix  is not running at each time-step. It's running before the simulation, and then restored afterward. My apologies.

Toby-D, 2017-02-13:
Haha. Well that's good to know too. 

By the way, I think that anything your group can do to document things like this before DOE funding runs out--how the code works and why it's implemented the way that it is--will go a long way in enabling others to maintain and develop the program.

Keep up the good work!

yuyihsiang, 2017-02-13:
Hi Toby

Yes, as Kelley pointed out, we have to do that for stability reason since added mass term is often significantly greater than the mass.

Regarding the mass actuator, I have never used it before. One thing I did notice by reading through the link you provided. I could be wrong but I think it is based on first generation of simMechanics, which may not be compatible with the second generation, which is what WEC-sim used. 

Thanks,

-Yi-Hsiang

Toby-D, 2017-02-14:
Thanks @yuyihsiang. And good point about the compatibility. I haven't looked into that.

kmruehl, 2017-02-20:
@Toby-D does this sufficiently address your issue?

Toby-D, 2017-02-20:
Yes. Thanks for the info!

kmruehl, 2017-02-20:
Glad we could help.

hachikoi1, 2023-03-14:
> 

Dear @kmruehl
    
Do you mean adjustMassMatrix works at the beginning of simulations and restoreMassMatrix works during post-processing, and process results are wrong? When I insert a Scope block to monitor total force in Hydrodynamic Body, I found the monitored total force is not the same as the results in the workspace.
So how can I get the right process results?
Best regards

kmruehl, 2023-03-30:
Refer to the discussion of this issue on https://github.com/WEC-Sim/WEC-Sim/issues/1006.

closed 2023-03-30 by kmruehl