WEC-Sim/WEC-Sim #162: Extract State Space for RM3 (A matric on vertical motion)

GuillaumeSab, 2017-02-13:
Dear WEC-Sim / Bemio team,

I am currently studying the RM3 WAMIT hydro data provided with WEC-Sim to derive an equation of motion model.
Usually, I use fitting approaches to identify radiation transfer function.

In the h5 file I saw that state space data are available.
I tried to extract it using h5read on Matlab but I cannot figure out which way should I do it to get the vertical motion related radiation transfer function from the 6*12*10*10 A matrix.

`rad_SS.A =h5read(file_h5,'/body1/hydro_coeffs/radiation_damping/state_space/A/all')`

I tried to get the corresponding coefficients this way but it is clear from the results that I am not getting the right system for the radiation:
`rad_SS.A_Z =rad_SS.A(:,:,3,3)`

Could you give me an hint on how to extract A matrix for the radiation function of vertical motion of the first body.

Thanks in advance,
Guillaume

j-vanrij, 2017-02-22:
Hi Guillaume,

You have it nearly correct.

For the RM3 .h5 data, Radiation_IRF_SS in bemio.m, was run with default values (maximum order or 10, and an R2 threshold of 0.95). For heave, for the first body (the float), the R2 threshold was obtained with a SS order of 2. You can extract the SS A, B, C, and D matrices with:

rad_SS.A =h5read('rm3.h5','/body1/hydro_coeffs/radiation_damping/state_space/A/all');
rad_SS.B =h5read('rm3.h5','/body1/hydro_coeffs/radiation_damping/state_space/B/all');
rad_SS.C =h5read('rm3.h5','/body1/hydro_coeffs/radiation_damping/state_space/C/all');
rad_SS.D =h5read('rm3.h5','/body1/hydro_coeffs/radiation_damping/state_space/D/all');
A = squeeze(rad_SS.A(:,:,3,3));
B = squeeze(rad_SS.B(:,:,3,3));
C = squeeze(rad_SS.C(:,:,3,3));
D = squeeze(rad_SS.D(3,3));

Thanks for the question!

Elise68, 2019-09-03:
Hi Bemio team,
I am currently trying to understand the state-space realisation. 
If I understood correctly, the state space realization is a method to solve the convolution integral term in the Cummins equation. Here is the approximation made:
[image]

Wec-Sim solves this equation with the time domain system identification based on Hankel Singular Value Decomposition.
Firstly, we obtain the equivalent state-space system Aij ,Bij, Cij, Dij. Using the above described methods, a set of state-space systems are obtained, one for each significant entry of the retardation matrix 𝐾. We obtain several state-space systems, according to:
 
[image]

where the subscripts 𝑖 and 𝑗 vary from 1 to 𝑚, where 𝑚 is the number of rigid-body platform degrees of freedom enabled (up to 6).
In order to obtain the complete state-space system, each of the matrices 𝐴𝑖𝑗, 𝐵𝑖𝑗 and 𝐶𝑖𝑗 have to be assembled into a global state-space system according to the following equations:
 
[image]
 
[image]

where 𝑛 describes the total number of radiation states, and 𝑚 the number of platform dof’s.

**Here are my questions about the Wec-Sim code:**
1.	The matrix A, B, C and D calculated in the code are the intermediate matrix (Aij, Bij, Cij) or the final matrix (Ar, Br, Cr)?
2.	Why is the A matrix a matrix of 4 dimensions and B/C matrix have 3 dimensions?
3.	I don’t understand, in your previous answer, why the vertical motion (heave) is linked to index 3 of the third dimension. Could you please explain this in more detail?

Thank you very much for your time and your answer.
Kind regards
Elise

nathanmtom, 2019-10-11:
@Elise68 Apologies for the delay in our response. The answers to your questions below:

1. For clarification the matrices A, B, C, and D correspond to the outputs from the code posted by j-vanrij in the prior response?
    a. If so, then the A, B, C, D matrices correspond to the intermediate matrix (Aij, Bij, Cij, Dij) the final  
    matrix (Ar, Br, Cr) can be found by running the RM3 simulation with simu.ssCalc = 1 and looking at 
    body(1).hydroForce.ssRadf. 
2. I am a little unclear about this question. If you were today, to run the script above the A matrix 
    would have nonzero entries for a in a 2 x 2 matrix in the upper left corner. The B and C matrices 
    have 2 non zero entries at the beginning of the vectors (1st two indices). However, perhaps I have       
    misunderstood your question.
3. If you were to execute "size(rad_SS.A)" in Matlab you should get "ans = 10 10 12 6". The first two 
   columns denote the default maximum matrix order of 10. The third column is 12 which is the 2 x 6 
   corresponding to the 6 degrees of motion of the two bodies, float and spar. The state space 
   matrices are also calculated for body to body interactions in BEMIO, but will not be used to 
   generate the final state space matrices unless the body-to-body flag is turned on. The final column 
   6 denotes the six degrees of freedom of body(1). Therefore, to obtain the A matrix for the heave 
   radiation force due to heave motion you will need to select the 3rd index of the third and fourth 
   dimensions. 

Please let me know if this has answered your questions and feel free to ask for clarification or follow up questions. 

Cheers, 
Nathan Tom
National Renewable Energy Laboratory

nathanmtom, 2019-10-30:
@Elise68 I wanted to check with you to see if our previous response answered your question? Please let us know if you have additional comments or questions.

Elise68, 2019-11-01:
Thank you very much for your answer.

Cheers,
Elise

closed 2019-11-01 by nathanmtom