WEC-Sim/WEC-Sim #168: mooring Stiffness

stevenshine, 2017-03-28:
hi
i  have  6 x 6 Stiffness matrix
how to set mooring line on floating  position?
[image]
example pic
that is set: 0 0 -13.15?
if the siffness matrix   set at two different positions   the results of displacement(RAO)   are different?
How to set mooring line position well?

yuyihsiang, 2017-04-25:
1. if the stiffness matrix set at two different positions the results of displacement(RAO) are different?
Yes, the results will be different
2. How to set mooring line position well? 
Please clarify your question.
3. What kind of condition the time period have Phase angle ?
Please clarify your question.

kmruehl, 2017-05-11:
@stevenshine does this help?

stevenshine, 2017-05-11:
ths u , i have solved.
i have another question
Are thre high order forces in WECSIM??
e.g:drift force.....

kmruehl, 2017-05-11:
No, we do not account for drift or current.

closed 2017-05-11 by kmruehl