WEC-Sim/WEC-Sim #170: Capability of WEC-Sim design tool to investigate a floating WEC without mooring line.

Krishnendu-Ocean, 2017-04-04:
Hello,

I came across the design tool WEC-Sim very recently. My area of research is in utilising the wave energy potential effectively. The tutorials provided were really interesting and informative. My study is mainly focused on experimental investigations and I am planning to consider a power take off system above the water level. However, I wanted to explore the possibility of numerical investigations.

Query,
Is it possible to test the response of a WEC system with the Power Take Off system/Heave constraint installed above the water level using WEC-Sim design tool? From the tutorial I could understand that this tool is capable of testing the floating body attached to the sea bed with mooring lines and a power take off system placed below the water level. What if a mooring less design have to be tested? Kindly let me know if WEC-Sim is capable of testing such systems.  

Thank you & Regards.

bradling, 2017-04-19:
Hello,

I think this wouldn't be a problem. Nothing in the PTO blocks or constraint blocks makes any assumption about being above or below water line. And there is no requirement to include a mooring block in your WEC-Sim model. You said your system has a heave constraint? Then a simulink model as shown below should meet your needs. 

I hope that is helpful for you.

[image]

Krishnendu-Ocean, 2017-04-20:
Hello,

Thanks for the reply.
If so I will go with the numerical simulations.

Thanks again.

Best Regards,

Krishnendu
Research Scholar
Ocean Engineering Division
Civil Engineering Department
IIT Bombay, Mumbai - 400076

closed 2017-04-25 by yuyihsiang

kmruehl, 2017-04-25:
@Krishnendu-Ocean, Brad's suggestion is appropriate. 
@bradling thanks for your reply on this one