WEC-Sim/WEC-Sim #171: Verification status for BEMIO function Read_AQWA

Toby-D, 2017-04-19:
WEC-Sim Team, 

Thanks for a good webinar yesterday on BEMIO and MCR. 

During the webinar, it was mentioned that the BEMIO function Read_AQWA.m has not been well verified.  Is that also true of the AQWA reader in the python version, or does only the MATLAB version warrant concern?

Thanks,

Toby.

j-vanrij, 2017-04-19:
Both the python and MATLAB versions of BEMIO were tested with the same, single, AQWA case (provided in the tutorials\BEMIO\AQWA\Example file). So, both the python and MATLAB versions of BEMIO have the same, minimal, level of verification. 

But to be clear, we have no knowledge that there _are_ bugs in Read_AQWA, and don’t want to cause concern, just thought it would be fair to warn users that Read_AQWA hasn’t been tested as extensively as Read_Nemoh and Read_WAMIT. 

So if you find issues please let us know. And thanks for attending the webinar and the feedback!

Toby-D, 2017-04-19:
Thanks for clarifying. 

We've used AQWA data in BEMIO, and the only bug that we discovered was fixed in WEC-Sim/bemio#37. We'll definitely let you know if we come across anything else.

closed 2017-04-25 by yuyihsiang