WEC-Sim/WEC-Sim #172: Nonlinear Froude-Krylov not changing with wave direction

TiagoMCA, 2017-04-28:
Hi,
I've noticed that, when using the nonlinear option in wec-sim, the froude-krilov pressures do not take into account the wave direction. This is easly seen with paraview for a nonzero wave direction (something like 90º is a good test). There are ways to overcome the issue, but I was wondering if its possible to correct this on the code?. Thanks for your time

yuyihsiang, 2017-05-04:
Hi TiagoMCA

Yes you are correct, in the current version, the the froude-krilov pressures do not take into account the wave direction. The wave directionality is something that we have not directly included in WEC-Sim. The easiest way to get around this is to rotate the device.

This is something on out todolist. Hopefully, we will be able to incorporate that in the near future.

Thanks,

-Yi-Hsiang

TiagoMCA, 2017-05-08:
Hi Yi-Hsiang

Sounds good to me. Thanks for the reply.

closed 2017-05-08 by kmruehl