WEC-Sim/WEC-Sim #173: Moordyn paraview export issue with dtOut

TiagoMCA, 2017-04-28:
Hi,
I've noticed that, when exporting moordyn results to paraview, the code doesn't take into account the simulation output timestep (dtOut). Thanks again for your time.

kmruehl, 2017-04-28:
@TiagoMCA thank you for bringing this issue to our attention, we will look into it.

bradling, 2017-05-09:
moorDyn supports changing the output sample rate, but you need to manually include your value for dtOut in in the moorDyn input file (mooring/lines.txt). See below:

`---------------------- SOLVER OPTIONS-----------------------------------------`
`0.01     dtOut        - output time step`
`0.001    dtM          - time step to` ...

@kmruehl is there any plan to use the github page to list wishlist tasks that contributors could perform going forward? This is something I would be willing to write a function to automate, but probably would not get to for a few weeks.

kmruehl, 2017-05-09:
@bradling that's a great idea.  I don't think the gh-pages site is the right place (because it has to be compiled each time), and the issues page isn't ideal for this either. Let me look into the best way to implement this, and I'll let you know.

kmruehl, 2017-05-09:
@bradling @TiagoMCA I looked into different places to host requested WEC-Sim feature additions, and it looks like the [WEC-Sim projects page](https://github.com/WEC-Sim/WEC-Sim/projects ) board is the best location. I went ahead and added a [project for automation of the dtOut from WEC-Sim into the MoorDyn input file](https://github.com/WEC-Sim/WEC-Sim/projects/4). This project board also provides a location to post the current status and related issue as well. I hope this helps. I'll update the WEC-Sim website to reflect this new board shortly.

closed 2017-05-09 by kmruehl