WEC-Sim/WEC-Sim #174: Permission to reproduce RM3 diagram from docs for paper

crobarcro, 2017-04-28:
Hello,

I am preparing a conference paper which makes use of WEC-Sim and would like permission to reproduce the diagram of the RM3 device from the documentation. I am citing your papers on WEC-Sim in the paper.

kmruehl, 2017-04-28:
sure, that's fine. let me know if you would like a better image, we have several.

crobarcro, 2017-04-28:
Yes, if you had a better image that would actually be really helpful thanks! A vector (svg or something like this) would be ideal, but just higher res would be great. Can you upload here or do you need an email address?

kmruehl, 2017-05-09:
All of the RM3 reports (and CAD model) are available here:
http://energy.sandia.gov/energy/renewable-energy/water-power/technology-development/reference-model-project-rmp/

What image are you looking for? Of the design or of its WEC-Sim model?

crobarcro, 2017-05-11:
In this case, I'm referring to the image below from the WEC-Sim online documentation:
[image]

crobarcro, 2017-05-11:
Incidentally, as the CAD files at the link you gave are in Solidworks format, they are not accessible to me, Solidworks costs many many thousands of pounds unfortunately. It would be nicer if they were available in a standard interchange format like STEP or IGES, although it's not an issue for me as I don't really need them and you provide the STL files in the WEC-Sim repo which is fine for my purposes, so just a comment really.

kmruehl, 2017-05-11:
Here is the PNG of that image:
[image]

Here are IGES files for the float and spar. I'll pass that feedback along to the Reference Model team about the file-type. It's a good suggestion.
[rm3_IGS.zip](https://github.com/WEC-Sim/WEC-Sim/files/993611/rm3_IGS.zip)

closed 2017-05-11 by kmruehl