WEC-Sim/WEC-Sim #175: Question of Radiation Force

10472008, 2017-05-10:
Hi,
I noted that "SS CI and Constant-Damping-CoeVariant Subsystem" block has 3 options ( Constant Coefficients、Convolution Integral Calculation and State-Space Calculation ) to calculate radiation damping force.

[image]

These options are used to calculate respectively in what situation?
(As far as I know ,in regular wave simulation the Constant Coefficient method will be taken to calculate radiation damping force.)
Thank you for your helping.
Kind Regards
Steve

kmruehl, 2017-05-10:
These variant subsystems are turned on/off depending on which parameters are defined in the `wecSimInputFile.m`. One way to see which subsystems are activated is to run the model and look at which blocks in the `*.slx  file are grayed out and which are active. 

Here's a general overview:
- **Constant Coeff** is used when `waves.type = noWave` or `regular`
- **CIC** is used when `waves.type = noWaveCIC`, `regularCIC`, and all irregular wave cases
- **SS** is used when 'simu.ssCalc = 1`

[Read more info about the waveClass here](http://wec-sim.github.io/WEC-Sim/code_structure.html#waveclass)

10472008, 2017-05-10:
To Kmruehl :
It is really helpful! Thank you very much!
Kind regards 
Steve

kmruehl, 2017-05-10:
This could be related to the quality of the BEM solution and the IRF calculation. Have you looked at you radiation damping and added mass terms at the period of interest?

10472008, 2017-05-11:
The Comparison of radiation damping and added-mass term, obviously radiation damping term look a little bit different.
[image]

Although Radiation term have some different, Excitation Force seems pretty well.
[image]

kmruehl, 2017-05-11:
Thanks for sharing the plots, this helps me understand the magnitude of the difference between CIC and Coeff for your case. Since this is a _relatively_ minor difference, I think it is likely due to the quality of the BEM solution and IRF calculation. I recommend looking closely at the frequency response of your BEM solution, specifically the frequency-domain Radiation Damping Coefficient, and the Radiation IRF. My guess is that (especially at the period of interest) there may be some instability in the BEM solution, or the IRF is not stable (ie not going to 0).

The BEMIO tutorials provide some examples along with plotting scripts. Below is an example from the NEMOH tutorial for the sphere. For this solution, the frequency-domain Radiation Damping Coefficient is unstable in heave around 2.2 rad/s, and for the IRF you can see that the SS solution is generally smoother and stable (ie settles to zero) quicker. 

[image]

[image]

These are things to check on the BEM side, prior to running WEC-Sim. Some useful references on how to do quality control of your bemio solution are available on the [BEMIO online documentation here](http://wec-sim.github.io/WEC-Sim/features.html#bemio), and with the [BEMIO tutorials here](https://github.com/WEC-Sim/WEC-Sim/tree/master/tutorials/BEMIO). However, admittedly the quality check of the BEM solution is something we do not thoroughly address, and could use more documentation. Hope this helps.

kmruehl, 2017-05-17:
@10472008 does this answer your question?

10472008, 2017-05-18:
Ms.Kelly Ruehl:
This does.Thanks a lot.
Sincerely,
Steve

closed 2017-05-18 by kmruehl