WEC-Sim/WEC-Sim #176: MoorDyn float buoyancy

mmacn, 2017-05-10:
Hello,

I have a MoorDyn question, probably for @mattEhall, but perhaps someone else can assist me. I am simulating a WEC with a mooring arrangement that has several surface piercing floats. In MoorDyn, it looks like the buoyancy of the floats does not stop at the free surface, but causes the mooring line to rise into the air. I'm not sure if I am setting up the MoorDyn input incorrectly, or if there is any way to simulate a mooring system with MoorDyn that doesn't exhibit this behavior?

I have attached a screenshot below.

[image]

Thanks very much for your help.

Michael

mattEhall, 2017-05-10:
Hi Michael,

You didn't miss anything. In fact MoorDyn doesn't currently consider the free surface at all. It was a feature that I thought risked causing more complications than it was worth at the time. Turning off buoyancy of float elements once they reach z=0 would be a quick fix that I could do, but it could lead to oscillation of floats at the water line. It raises the next question of modelling the hydrostatic stiffness of the float ... I guess we could stick with the spherical assumption for that. Then the next issue comes: the surface elevation should be changing according to the waves, which aren't currently handled in MoorDyn. You can see the difficulty.

My current thought for what's best in these situations (with MoorDyn in general, not specific to WEC-Sim) is to use MoorDyn for only the moorings and simple submerged floats. It'd be best to model anything with more involved hydrodynamics (such as a surface-piercing float) with whatever hydrodynamics model is coupled with MoorDyn. MoorDyn currently has the capability for that type of multi-body coupling, but we haven't worked it out for the MoorDyn-WEC-Sim coupling yet. I think that would be the only really good solution for you.

Sorry I don't have a decent quick fix.

- Matt

kmruehl, 2017-05-11:
@mattEhall Thank you for this detailed response. 

Following up with Matt's suggestion to model the mooring with a combo of WEC-Sim/MoorDyn, you could model the surface piercing floats in WEC-Sim and apply a stiffness matrix between them and the spar (or you could apply a user-defined force if you know the non-linear relationship between the surface float and spar). Then, as Matt suggested, you could model the remaining mooring system in MoorDyn, ie the connection from the surface piercing floats to the seafloor. 

Hope this helps!

closed 2017-05-11 by kmruehl

mmacn, 2017-05-11:
Thanks @mattEhall and @kmruehl very much for your suggestions!