WEC-Sim/WEC-Sim #177: Problem calling wecSimMCR twice

jo4c, 2017-05-15:
Hi,

I have some code set up to run a set of WEC-Sim runs, then do a second pass to increase the level of detail around points of interest.  The top level m-file has a structure like this:
```
[other code]
write mcrCaseFile
write wecSimInputFile.m
wecSimMCR
post-process results

determine new cases to run
write new mcrCaseFile
write new wecSimInputFile.m (referring to new mcrCaseFile)
wecSimMCR
post-process results
```
The problem I have is that wecSimMCR does not appear to load the new mcrCaseFile.m, but instead re-uses the original one.  It is as if it if loaded in a cache somewhere and not cleared in between.  Is there a way to make it force reload mcrCaseFile.m so that it will run the new set of runs?

This behaviour appears to occur _**only**_ when running through the entire block of code above in one go.  That is, if I step through line-by-line (or even 1st block of code above, then 2nd block) then it behaves as expected, using the 2nd pass mcrCaseFile.  Any ideas?

Kind regards,
Jo

bradling, 2017-05-15:
I have run into weird things like this before. I think what is happening is matlab is saving the script "wecSimInputFile.m" when you run the script, and then running it from memory. It does not reread the script between calls.

Try putting a `clear wecSimInputFile` statement before calling `wecSimMCR` the second time. I think that might do the trick.

jo4c, 2017-05-16:
Thanks very much @bradling - that seems to have done the trick.
(I had tried clearing various other properties previously, but to no avail - wasn't aware that the input file itself got stored.)

closed 2017-05-16 by jo4c

kmruehl, 2017-05-17:
@bradling thanks for answering this!