WEC-Sim/WEC-Sim #179: Error in simulating two OSWECs

drschandel, 2017-05-16:
Dear WEC-Sim team

I am encountering an error when trying to simulate a pair of OSWECs in parallel to each other. I'd really appreciate some help on this because I've been stuck on it for a while. I have double checked that the coefficients from NEMOH have been calculated properly for 4 bodies, all numbered in the same order as the WEC-Sim model.

It works fine for a single OSWEC, When I try to run wecSim for two devices, the output looks like this:

> WEC-Sim Read Input File ...   
> Elapsed time is 24.595712 seconds.
> 
> WEC-Sim Pre-processing ...   
> Elapsed time is 0.419894 seconds.
> 
> WEC-Sim Simulation Settings:
> 	Time Marching Solver                 = Fourth-Order Runge-Kutta Formula 
> 	Start Time                     (sec) = 0
> 	End Time                       (sec) = 100
> 	Time Step Size                 (sec) = 0.01
> 	Ramp Function Time             (sec) = 10
> 	Total Number of Time Step            = 10000 
> 
> Wave Environment: 
> 	Wave Type                            = Regular Waves (Sinusoidal Steady-State)
> 	Wave Height H (m)                    = 0.1
> 	Wave Period T (sec)                  = 3
> 
> List of Body: Number of Bodies = 4 
> 
> 	***** Body Number 0, Name: mesh1 *****
> 	Body CG                          (m) = [0,0,-0.528]
> 	Body Mass                       (kg) = 6.1 
> 	Body Diagonal MOI              (kgm2)= [103,103,103]
> 
> 	***** Body Number 1, Name: mesh2 *****
> 	Body CG                          (m) = [-1.2728,0.0728,-0.528]
> 	Body Mass                       (kg) = 6.1 
> 	Body Diagonal MOI              (kgm2)= [103,103,103]
> 
> 	***** Body Number 2, Name: mesh3 *****
> 	Body CG                          (m) = [0,0,-0.578]
> 	Body Mass                       (kg) = 999 
> 	Body Diagonal MOI              (kgm2)= [999,999,999]
> 
> 	***** Body Number 3, Name: mesh4 *****
> 	Body CG                          (m) = [-1.2728,0.0728,-0.578]
> 	Body Mass                       (kg) = 999 
> 	Body Diagonal MOI              (kgm2)= [999,999,999]
> 
> List of PTO(s): Number of PTOs = 2 
> 
> 	***** PTO Name: PTO1 *****
> 	PTO Stiffness           (N/m;Nm/rad) = 0
> 	PTO Damping           (Ns/m;Nsm/rad) = 200
> 
> 	***** PTO Name: PTO2 *****
> 	PTO Stiffness           (N/m;Nm/rad) = 0
> 	PTO Damping           (Ns/m;Nsm/rad) = 200
> 
> List of Constraint(s): Number of Constraints = 2 
> 
> 	***** Constraint Name: Constraint1 *****
> 
> 	***** Constraint Name: Constraint2 *****
> 
> 
> Simulating the WEC device defined in the SimMechanics model OSWEC_double.slx...   
> Error using wecSim (line 217)
> Error due to multiple causes.
> 
> Caused by:
>     Error using wecSim (line 217)
>     Error evaluating parameter 'RotationMatrix' in 'OSWEC_double/Rotational PTO/Rigid  Transform1'
>         Error using wecSim (line 217)
>         Expected one output from a curly brace or dot indexing expression, but there were 2 results.
>     Error using wecSim (line 217)
>     Error evaluating parameter 'RotationMatrix' in 'OSWEC_double/Rotational PTO1/Rigid  Transform1'
>         Error using wecSim (line 217)
>         Expected one output from a curly brace or dot indexing expression, but there were 2 results.`

The error seems to result from the very first 'Rigid Transform 1' block under pto(1):

[image]

The simulink model  has been adapted straight from the single OSWEC tutorial.

My Wec-Sim input file looks like this:

> %% Simulation Data
> simu = simulationClass();
> simu.simMechanicsFile = 'OSWEC_double.slx';          % Specify Simulink Model File
> simu.mode = 'normal';                       % Specify Simulation Mode 
> %                              ('normal','accelerator','rapid-accelerator')
> simu.explorer='on';                         % Turn SimMechanics Explorer (on/off)
> % simu.startTime = 0;                         % Simulation Start Time [s]
> simu.endTime=100;
> simu.dt = 0.01;
> simu.rampT = 10;
> simu.CITime = 7.5;
> 
> %% Wave Information
> 
> %Regular Waves 
> waves = waveClass('regular');
> waves.H = 0.1;
> waves.T = 3;
> 
> % %Irregular Waves using PM Spectrum
> % waves = waveClass('irregular');
> % waves.H = 2.5;
> % waves.T = 8;
> % waves.spectrumType = 'PM';
> 
> % %Irregular Waves using User-Defined Spectrum
> % waves = waveClass('irregularImport');
> % waves.spectrumDataFile = 'ndbcBuoyData.txt';
> 
> %% Body Data
> body(1) = bodyClass('hydroData/Nemoh.h5');   % Initialize bodyClass for Flap
> body(1).mass = 6.1;                         % User-Defined mass [kg]
> body(1).momOfInertia = [103 103 103]; % Moment of Inertia [kg-m^2]
> body(1).geometryFile = 'geometry/flap_translated_1.stl';%translated_1.stl'%'geometry/flap_translated.stl';    % Geometry File
> 
> body(2) = bodyClass('hydroData/Nemoh.h5');   % Initialize bodyClass for Flap
> body(2).mass = 6.1;                         % User-Defined mass [kg]
> body(2).momOfInertia = [103 103 103]; % Moment of Inertia [kg-m^2]
> body(2).geometryFile = 'geometry/flap_translated_1.stl';%translated_1.stl'%'geometry/flap_translated.stl';    % Geometry File
> 
> body(3) = bodyClass('hydroData/Nemoh.h5');   % Initialize bodyClass for Base
> body(3).geometryFile = 'geometry/base_translated_1.stl';    % Geometry File
> body(3).mass = 'fixed';                        % Creates Fixed Body
> 
> body(4) = bodyClass('hydroData/Nemoh.h5');   % Initialize bodyClass for Base
> body(4).geometryFile = 'geometry/base_translated_1.stl';    % Geometry File
> body(4).mass = 'fixed';    
>     
> %% PTO and Constraint Parameters
> constraint(1)= constraintClass('Constraint1'); % Initialize ConstraintClass for Constraint1
> constraint(1).loc = [0 0 -0.578];
> 
> constraint(2)= constraintClass('Constraint2'); % Initialize ConstraintClass for Constraint1
> constraint(2).loc = [-1.2728 0.0728 -0.578];
> 
> pto(1) = ptoClass('PTO1');                     % Initialize ptoClass for PTO1
> pto(1).k = 0;                                  % PTO Stiffness Coeff [Nm/rad]
> pto(1).c = 200;                                  % PTO Damping Coeff [Nsm/rad]
> pto(1).loc = [0 0 -0.528];                       % PTO Global Location [m]
> 
> pto(2) = ptoClass('PTO2');                     % Initialize ptoClass for PTO1
> pto(2).k = 0;                                  % PTO Stiffness Coeff [Nm/rad]
> pto(2).c = 200;                                  % PTO Damping Coeff [Nsm/rad]
> pto(2).loc = [-1.2728 0.0728 -0.528];                       % PTO Global Location [m]

I'd really appreciate some help or comments. Thanks

kmruehl, 2017-05-17:
Would you be willing to attach a *.zip of the project file so that I can troubleshoot? I don't see anything that should have caused this error based on what you've posted. Alternatively, you could email it to me at kelley.ruehl@sandia.gov

Thank you!

drschandel, 2017-05-30:
Hi Kelly

Thanks for the reply. I emailed you the project file, but maybe you didn't receive it. Here it is again:

[doubleWEC.zip](https://github.com/WEC-Sim/WEC-Sim/files/1037802/doubleWEC.zip)

kmruehl, 2017-06-07:
@drschandel Thank you so much for uploading the *.zip, and sorry for the delay. I was able to run the model as it and had no problems. I'll send my model back to you. Have you pulled the latest version of WEC-Sim and set your path correctly? Refer to #181 for path information

drschandel, 2017-06-08:
Hi, thanks for the reply. I updated my WEC-Sim and added that variable and it worked like a charm!

I think I was doing 'clear all' somewhere in my script and that was the problem. 

I really appreciate you guys building this great tool and providing support for it. Cheers!

kmruehl, 2017-06-08:
@drschandel glad that fixed the problem and that you enjoy the code :)

closed 2017-06-08 by kmruehl