WEC-Sim/WEC-Sim #182: RM3 coupled with MoorDyn error

arash1988a, 2017-06-07:
Hello everybody,

I try to run RM3_MoorDyn tutorial and I get this error:

<Error using wecSim (line 236)
Error evaluating 'InitFcn' callback of SubSystem block 'RM3MoorDyn/MoorDyn/MoorDyn/MoorDyn'.

Caused by:
    Error using wecSim (line 236)
    C:\...\AppData\Local\Temp\tp13dbe15c_7c8e_41e8_b238_3abe9f546b41\Lines_thunk_pcwin64.dll is not a
    valid Win32 application.>

I have MoorDyn.h file in the right directory, and I run it on Matlab2015b. Can anyone help me how to run it?

Thanks

yuyihsiang, 2017-06-07:
Hi 

We have just tested the MoorDyn run and it was working fine on both Windows and OS X.

It seems that you may have a C compiler issue. Please  check and see if you have installed MinGW. If you don't, you can install it by clicking the "Add-on" under the Home tab.

https://www.mathworks.com/help/matlab/matlab_external/install-mingw-support-package.html

-Yi-Hsiang

yuyihsiang, 2017-06-07:
I also tested several different MATLAB versions from 2015b to 2017a. As long as the C compiler is installed properly WEC-SIm/MoorDyn works fine.

arash1988a, 2017-06-07:
Hello dear Yuyihsiang

Before installing MinGW compiler it gave me different error, I installed it
then I cheked with the mex command "mex -setup" and it seems that the
compiler has been installed, is it possible that there's still something
wrong with the compiler? How can I make sure that the problem comes from
that?
Thanks

yuyihsiang, 2017-06-07:
It will be great if you can check and make sure you have the latest version of WEC-Sim and the RM3_MoorDyn from WEC-Sim Applications.

It will be also helpful to let us know which OS system you are using.

Thanks,

-Yi-Hsiang

arash1988a, 2017-06-07:
I cloned the latest version of Wecsim and its applications from repository
of github, I'm using windows 10

arash1988a, 2017-06-07:
Windows 10 64bit

yuyihsiang, 2017-06-07:
Hum, I have never tried to run it on a Windows 10 machine, which could be a problem for the compiler.

My suggestion will be:
1. One think you can try is to upgrade to MATLAB 2016a or newer.
or
2. Remove the c compiler and reinstall it to make sure it is fine. 
 
-Yi-Hsiang

arash1988a, 2017-06-07:
Thank you very much, I will do as you say and update you with the results

arash1988a, 2017-06-08:
Dear Yuyihsiang

I uninstalled the compiler and downloaded it again and finally MoorDyn
worked, even with windows10. Thanks for your help

On Wed, Jun 7, 2017 at 11:24 PM, Arash Alikhani <arash1988a@gmail.com>
wrote:

> Thank you very much, I will do as you say and update you with the results
>

yuyihsiang, 2017-06-08:
Great,

Thanks,

arash1988a, 2017-06-09:
Dear Yuyihsiang

Finally I could run the RM3_Moordyn coupled successfully, but for any other
case I can't run my WEC device with moordyn, it suddenly crashes. Have any
idea?

yuyihsiang, 2017-06-09:
Please upload your case folder (zipped) here and I can take a look to see what happen. It could be that the moorDyn setting need to be changed or something else.  

-Yi-Hsiang

arash1988a, 2017-06-12:
Good morning,

1st attachment: the results visualization of RM3 coupled with MoorDyn in
paraview just to make sure that everything works well
2nd attachment: our device simulation without mooring to make sure that
there's no error regarding this simulation
3rd attachment: wecsim input file adding MooringClass
4th attachment: simulink file with MoorDyn
5th attachment: All the folders updated and a list of them shown here
6th attachment: MoorDyn input file
7th attachment: The error without any explanation! The program just crash

kmruehl, 2017-06-13:
@arash1988a I don't see an attachment

arash1988a, 2017-06-14:
Dear Kelley, I attached them again, If still they are not visible I cand
send them directly to your mail.

kmruehl, 2017-06-14:
@yuyihsiang do you see a zip file? I still don't.

yuyihsiang, 2017-06-19:
@arash1988a I also did not see any attachment. How large is the file?

-Yi-Hsiang

arash1988a, 2017-06-20:
I uploaded them on google drive and here is the shared link:

https://drive.google.com/open?id=0B_DLeq5D_oaLN1Uxd3lnT2ZTWlE

kmruehl, 2017-06-29:
@arash1988a I downloaded and viewed your attachment with the error messages, however without the WEC-Sim model itself, I cannot troubleshoot the error. If this is something you're willing to share, can you ZIP and upload the case directory of this model (ie: input file, simulink file, etc)?

kmruehl, 2017-06-29:
Have you contacted Mathworks directly? This looks more like a Mathworks/Compiler issue, not necessarily a WEC-Sim/MoorDyn issue.

kmruehl, 2017-07-17:
@arash1988a did this resolve your issue?

arash1988a, 2017-07-18:
Thanks Kelly, yes I resolved the problem and ran the simulation, then I
encountered another problem. The simulation runs successfully but when I
try to write a paraview file to visualize the output, the mooring line
can't be seen there, even though I had almost 10 segments on it, the only
visible thing is two dots at the bottom surface.

kmruehl, 2017-08-02:
@arash1988a We just did a webinar on Mooring/Viz [available here](http://wec-sim.github.io/WEC-Sim/webinars.html#wec-sim-webinar-4-mooring-and-visualization). Unfortunately the audio on the visualization webinar is terrible and needs to be re-recorded. I plan to do that today and upload the recording. 

In the meantime the slides are available, and I've uploaded an RM3 coupled with MoorDyn example using ParaView for visualization to the [Applications repository here](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Viz/RM3_MoorDyn_Viz). I think What you need to do is increase the number of `moorDynNodes` used for visualization in the wecSimInputFile, as shown below:
`mooring(1).moorDynNodes = [21 21 21];       	% Specify number of nodes per line`

arash1988a, 2017-08-03:
Thanks dear Kelley, it was really helpful

kmruehl, 2017-08-04:
Glad to help/

closed 2017-08-04 by kmruehl