WEC-Sim/WEC-Sim #184: WEC shape study using WEC-SIM

nikdave14, 2017-06-08:
Hello,

It has been only three weeks that I started working on WEC-SIM using MATLAB R2016b. I found it interesting and have successfully completed the tutorials on RM3 and OSWEC. The tutorials are comprehensive and well explained. I started using MATLAB/programming tools very recently, so you might find the questions very stupid. Apologies for that!
I am a student and the task is to study the effect of using different basic shapes (like cylinder, sphere, cube etc.) of single body WEC. The floating shape is connected with a mooring cable with the seabed. I started with a vertical cylinder case (without mooring) which I could simulate (not sure if the results are physical though!), which has heave and surge motion. Now, I did the tutorial on the website, RM3 using MoorDyn, and tried reflecting the same with a basic cylindrical shape. The results for the RM3 (without mooring) tutorial and RM3 with MoorDyn are exactly same if we add an initial displacement of -0.21 m in the z direction (as in the tutorial). 
Reflecting the same mooring (no. of cables 6, using same lines.txt file and parameters), I could simulate the cylinder, which now just has heave motion. But now, when I use only one mooring cable, I get this error:

Error using wecSim (line 237)
Derivative of state 'Cylinder.Translational_PTO.Translational_PCC_Prismatic_Joint.Pz.v' in block
'Cylinder/Translational PTO/Translational PCC (Prismatic Joint)' at time 48.900000000000006 is not finite. The simulation will be stopped. There may be a singularity in the solution.  If not, try reducing the step size (either by reducing the fixed step size or by tightening the error tolerances) 

I think this might be because of some problem in my simulink model. I can share the files, I am using. Please let me know how can I do that. You can contact me at nikdave14@gmail.com.

Am I on the right track for solving a basic shape WEC with single mooring cable? I have just four weeks left to present this study. Could you please refer me some papers or presentations, regarding this study done in the past to validate/verify my results.

Many thanks!
Nik

kmruehl, 2017-06-08:
@nikdave14 Thanks for the detailed explanation of your work. The error you are receiving is due to the time-step you have selected. You can resolve this by making your time-step smaller. This is done by setting `simu.dt`. We recommend making your time-step approximately T/100 or T/200, where T is the incident wave period `waves.T`. Hope this helps.

nikdave14, 2017-06-09:
@kmruehl 
Thanks for your help Kelley. Appreciate it.
The incident wave period for my simulation, waves.T = 300 and time step size, simu.dt = 0.01. I tried reducing it further, but still the above error message pops up. Could this be related to my simulink model (PFA image).
[image]

 I am not sure where exactly should I connect the PTO link in a single body case with mooring. In the RM3 tutorial, the PTO is connected between the two bodies.

Also, how can I validate my results for the WEC shape study?

Cheers,
Nik

nikdave14, 2017-06-12:
@yuyihsiang , many thanks for your help. 
I removed Floating DOF constraint from both the simulink model and input file. 
I am attaching an image of both below,

[image]
[image]

But I get this error now,

WEC-Sim Read Input File ...   
Elapsed time is 0.073531 seconds.
WEC-Sim Pre-processing ...   
Elapsed time is 0.113801 seconds.
WEC-Sim Simulation Settings:
	Start Time                     (sec) = 0
	End Time                       (sec) = 400
	Time Step Size                 (sec) = 0.01
	Ramp Function Time             (sec) = 100
	Convolution Integral Interval  (sec) = 5
	Total Number of Time Step            = 40000 
Wave Environment: 
	Wave Type                            = Regular Waves (Convolution Integral Calculation)
	Wave Height H (m)                    = 2.5
	Wave Period T (sec)                  = 300
List of Body: Number of Bodies = 1 
	***** Body Number 0, Name: axisym *****
	Body CG                          (m) = [0,0,-1]
	Body Mass                       (kg) = 3.5337E+06 
	Body Diagonal MOI              (kgm2)= [2.09073E+07,2.13061E+07,3.70855E+07]
List of PTO(s): Number of PTOs = 1 
	***** PTO Name: PTO1 *****
	PTO Stiffness           (N/m;Nm/rad) = 0
	PTO Damping           (Ns/m;Nsm/rad) = 1.2E+06
List of Constraint(s): No Constraint in the system
Simulating the WEC device defined in the SimMechanics model Cylinder.slx...   
Error using wecSim (line 237)
Error evaluating parameter 'TranslationCartesianOffset' in 'Cylinder/Translational PTO/Rigid  Transform
(Orientation)'
Caused by:
    Error using wecSim (line 237)
    Undefined variable "constraint" or class "constraint.loc".

Cheers,
Nik

yuyihsiang, 2017-06-12:
@nikdave14 It seems to me that you are using an older WEC-Sim Lib. There was an initialization bug in WEC-Sim PTO blocks. The code was looking for constraint parameters in the PTO block, which is not right. The bug was fixed, a few month ago. Please pull the latest one and let me know if it works.

Thanks,

--Yi-Hsiang

kmruehl, 2017-06-12:
@yuyihsiang thanks for responding to this question. 

@nikdave14, I agree with everything Yi-Hsiang has said. I would also recommend that you check your BEM data, and make sure that you have hydro data for a 300s wave period (`waves.T = 300`). Also, you'll want to look into your simulation settings. You currently have:
```
simu.startTime =0;
simu.rampT = 100;                  
simu.endTime=400;                    
simu.dt = 0.1; 							
waves.T = 300; 
```
Which means that you are only simulating one complete wave period after the wave ramp. Is this your intent?

nikdave14, 2017-06-13:
@yuyihsiang , thanks for your help. I pulled the latest one and still receiving the same error.

@kmruehl , thanks, I changed the wave period wave.T after looking at the hydrodynamic data. Please find below image,
[image]

Thanks for suggesting the simulation settings. This is not intended, so I should change the simu.endTime to may be 800-1000. Right.

Cheers,
Nik

yuyihsiang, 2017-06-13:
In this case, you actually don't need a mooring because you have a single body system and it only moves in heave. You should remove the mooring block and associated inputs in the input file.

Running wave.T=300 is strange to me, even if the value is within the range of your hydrodynamics data. It is hard to find cases where you have a wave period of 300 sec, unless your are think about Infragravity waves or something else. If that is the case, your end time should be significantly longer than your wave period. You may also need a longer ramp time.

Thanks,

-Yi-Hsiang

nikdave14, 2017-06-13:
@yuyihsiang, @kmruehl  Thank you guys! This is very helpful. I will do the required changes for the vertical cylinder case. Now, as per my understanding, for calculating the efficiency of the shape, I need to find the ratio of absorbed power and available power, given as:
[image]
wherein in frequency domain,
[image]

and,
[image]
where, Response Amplitude Operator (RAO) is required,
[image]

Then in time domain,
[image]

Now, for available power,
[image]

where,
[image]
and,
[image]

Problems:
1)  I am finding it difficult to understand how to get the RAO from my Nemoh Results, which will be used to calculate the absorbed power, and is also useful to calculate drift forces (separate analysis). Is my understanding correct? Please correct me and suggest a way to calculate the RAO / efficiency for comparing different single body WEC shapes.

2) How could I verify the results received for the efficiency of a basic WEC shape. Have you done any similar study previously? Any results, I could compare to for simple shapes like sphere, cylinder etc.  

Cheers,
Nik

yuyihsiang, 2017-06-19:
Hi Nik

The RAO should be calculated from WEC-Sim simulation. For heave only case, I would run it with a wave height of 2m (amplitude of 1m) in regular waves (with a range of wave frequencies) and calculate the response amplitude for each wave frequency. For heave only case with linear hydrodynamics, there is actually no drift force. The same approach can be applied for different body shape.

I think you can compare with out IEA OES paper. Please note this is not the final version. please do not disseminate. The final version with come out in Aug.
[iea-oes-task.pdf](https://github.com/WEC-Sim/WEC-Sim/files/1086488/iea-oes-task.pdf)

Thanks,

-Yi-Hsiang

closed 2017-06-29 by kmruehl