WEC-Sim/WEC-Sim #185: Limited movement

AleixWG, 2017-06-16:
Hello,

I'm Aleix again, we made great advances thanks to your previous suggestions. We succesfully added our own PTO and negative spring models in wecsim, but we are facing a new challenge which i hope you can help us with.

This is a 2-body WEC (floater and spar), the floater can move in heave but not indefinitely, it has one upper limit and one bottom limit, how can we introduce this limits in the simulink model?

Thank you

kmruehl, 2017-06-19:
@AleixWG the best suggestion we have for implementing end-stops is to add a _very_ stiff spring that activates at a certain distance. To the best of our knowledge, the Simscape joints do not offer a way to directly implement end-stops, so this is the recommended approach.

kmruehl, 2017-06-19:
@AleixWG I looked into it a bit more, and it looks like Simscape includes both a [Translational Hard Stop ](
http://www.mathworks.com/help/physmod/simscape/ref/translationalhardstop.html) and a [Rotational Hard Stop](http://www.mathworks.com/help/physmod/simscape/ref/rotationalhardstop.html) block that do the same thing.

AleixWG, 2017-06-23:
Thanks for the feedback, we are exploring this transitional hard stop and how to set it in the PTO joint with no luck so far but we will make it eventually.

kmruehl, 2017-06-26:
Let us know if you make any progress or have any more questions.

closed 2017-06-26 by kmruehl

vanderJan, 2019-09-29:
Hi,

Thank you for your information. it makes much sense for me to understand the implementation of endstop force. But I still can not make it according to your explanation as I am new to wecSim. It would be more than helpful if you can give me more information (the better, if there a picture of such a simplified simulink model ) about how to implement the very stiff spring when position exceed certain point.  or to be specific, how to judge the position?

many thanks!