WEC-Sim/WEC-Sim #186: AQWA excitation phase

kmruehl, 2017-06-22:
I am trying two identical cylinders separated by 5 diameters in the x direction and I get the attached.  I am most interested in heave motion.  The body 2 results seem a little bonkers, but generally in phase.  Should I really consider switching to WAMIT, or is there some way I can get around it with what I have from AQWA?
-Bret

kmruehl, 2017-06-22:
Does that excitation force include both Froude-Krylov and diffraction? If it does, can you plot only the Froude-Krylov? The Froude Krylov force should be identical for both bodies except for a phase shift. Also it would help if you can plot wave length in the x-axis instead of frequency. Also, what is the diameter of the cylinders?
-Carlos

kmruehl, 2017-06-22:
Thanks for the questions.  It turns out that the Froude-Krylov values are all NaNs.  I checked the h5 file and they are nans there too.  I peeked at the .lis file and things look OK there.  The diameter of the cylinders is 0.62m.  I am modeling at tank test scale.  Any suggestions?
-Bret

kmruehl, 2017-06-22:
Sounds like WEC-Sim has a bug when loading the lis file 
-Carlos

kmruehl, 2017-06-22:
In Read_AQWA.m for BEMIO the following is near the end of the script.
```
% Eventually update this to read in excitation force components from .LIS file here
hydro(F).sc_ma = NaN(size(hydro(F).ex_ma));
hydro(F).sc_ph = NaN(size(hydro(F).ex_ph));
hydro(F).sc_re = NaN(size(hydro(F).ex_re));
hydro(F).sc_im = NaN(size(hydro(F).ex_im));
hydro(F).fk_ma = NaN(size(hydro(F).ex_ma)); 
hydro(F).fk_ph = NaN(size(hydro(F).ex_ph));
hydro(F).fk_re = NaN(size(hydro(F).ex_re));
hydro(F).fk_im = NaN(size(hydro(F).ex_im));
```

-Bret

dforbush2, 2017-08-24:
The Read_AQWA script has been updated and now reads in Froude-Krylov+diffraction (.fk) and diffraction (.sc) from the .lis output file.

kmruehl, 2017-08-24:
@dforbush2 thank you!

kmruehl, 2017-10-19:
issue from @bretbosma

I have two heaving cylinders spaced with a one diameter gap between them offshore (x direction).  I have run wecSim and am plotting the resultant heave positions of both bodies as shown below.  The amplitude response is different as I might expect, but the phase seems to be the same.  If the wave is propagating through it seems to me that there should be a phase difference.

 [image]

If I understood our conversation from before you indicated that the phase shift should be included in the excitation force.  I have plotted that below and don't see a phase shift.
[image] 

Please let me know if  I am thinking about this wrong and if you have any suggestions of things to try.  Does this work differently for the other BEM inputs?

kmruehl, 2017-11-14:
@bretbosma, thank you for your patience. 

@dpatterson-nmt @dforbush2 and I have looked into this issue with the AQWA excitation phase rather extensively. While we agree that the excitation phase produced by the WEC-Sim response above appears incorrect (in phase when it should be out of phase), we've compared the direct AQWA output (from the *.lis file) with the output from BEMIO parser (in the *.h5 file) for several multi-body cases, and the excitation phases from all the cases we checked match. This leads us to believe is that issue is with how AQWA reports excitation phase, not with how BEMIO or WEC-Sim parse the excitation phase. 

There may be a flag in AQWA that reports the excitation phase in global coordinates instead of body coordinates (or similar), but without access to AQWA, we cannot further debug the issue. In short, we recommend using another BEM solver (WAMIT or NEMOH) when modeling arrays of WECs to ensure the excitation phase is properly reported. 

Attached is an excel sheet which plots the AQWA output directly from the *.lis file, and below are plots comparing the excitation phase from the direct AQWA output with the BEMIO excitation phase . 
  
**Plot of AQWA (*.lis) output**
[image] [image] [image]

**Plot of AQWA BEMIO parser (*.h5) output**
[image]

**Excel sheet plotting the AQWA (*.lis) output**
[X_compare.xlsx](https://github.com/WEC-Sim/WEC-Sim/files/1472583/X_compare.xlsx)

closed 2017-11-14 by kmruehl