WEC-Sim/WEC-Sim #187: Acceleration loop

AleixWG, 2017-06-29:
Im modelling a 2-body WEC + negative spring, for that im using a "Transitional PTO Actuation Force", the force i use as input is the sum of:

- Negative spring force
- Custom PTO force

For that forces i use position, velocities and accelerations of each body (response).

When i run the simulation without the acceleration i dont have any kind of problem, but when i try to run it with the acceleration links enabled the model find loops with the solver configurator.

The acceleration is not a big issue for the negative spring, but the custom PTO need the relative acceleration, any ideas how can i solve this loop problem?

Thanks,
Aleix

AleixWG, 2017-06-29:
Nevermind, i fixed that with a simple memory block in each acceleration.

kmruehl, 2017-06-29:
@AleixWG that's what I would have suggested. I'm guessing you got the algebraic loop error? If you look at how the added mass is implemented in the body block, you'll see that we also had to implement a work-around to handle the acceleration.

closed 2017-06-29 by kmruehl