WEC-Sim/WEC-Sim #188: Running examples - Error using h5readc

aritradatta, 2017-07-01:
We are trying to run the examples but we are running into the error below when we run the 'wecSim' command:

WEC-Sim Read Input File ...   
Error using h5readc
Unable to open file. Filename may be corrupt or have unsupported characters.

Error in h5read (line 58)
[data,var_class] = h5readc(Filename,Dataset,start,count,stride);

Error in bodyClass/readH5File (line 81)
            obj.cg = h5read(filename,[name '/properties/cg']);

Error in wecSim (line 85)
        body(ii).readH5File;

We noticed that in the readme file it is mentioned that "The *hf5 needs to be updated from BEMIO to include StateSpace" for the RM3 example. Can you please help us understand the steps involved? We are relatively new to this software. Any help will be greatly appreciated!

aritradatta, 2017-07-02:
EDIT after doing some debugging it seems that matlab is having an issue with opening oswec.h5 .. We have tried to run matlab as admin and that didnt help.

kmruehl, 2017-07-05:
I just ran the OSWEC tutorial and had no issues loading the oswec.h5 files. I've attached the file i ran. What version of MATLAB are you running? Are you sure you have the *.h5 file in the correct path? The ReadMe note is only relevant if you're running WEC-Sim with state space, it will not cause an error when importing the *h5 file to matlab. 

[oswec.zip](https://github.com/WEC-Sim/WEC-Sim/files/1125319/oswec.zip)

aritradatta, 2017-07-05:
@kmruehl we found the issue - we downloaded the repo from GitHub as a zip but for some reason the h5 files didn't download properly. For some reason we ended up with a ~1KB file with a Git hash, instead of the original ~7MB file. Replacing the file correctly solved our issue! Apologies for the confusion and thank you for your help :)

closed 2017-07-05 by aritradatta

kmruehl, 2017-07-05:
I'm glad you were able to resolve this issue. We've taken note of this, and will add a README (or similar) to remind users to check their H5 files.