WEC-Sim/WEC-Sim #191: ode14x compatability?

bradling, 2017-07-20:
Hi,

I am trying to simulate a PTO model using simscape, and my resulting physical model has an algebraic loop. Simulink recommends using the implicit solver ode14x in this case, an implicit fixed time step solver. Is this solver compatible with WEC-Sim?

kmruehl, 2017-07-31:
@bradling I have not tested WEC-Sim with `ode14x`, I've only done comparisons between`ode4 `and `ode45 `(see [documentation here](http://wec-sim.github.io/WEC-Sim/features.html#time-step-features) and [application here](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/NonlinearHydro)). I recommend running your model with the default WEC-Sim solver (currently `ode4 `- a fixed time-step solver) and comparing it to the `ode14x` solution. Since you have an algebraic loop when the model is coupled with the PTO, consider doing the comparison without the PTO model. Another good check would be to run `ode14x` with the [verification cases](https://github.com/WEC-Sim/WEC-Sim/tree/master/tutorials/verificationCases).

bradling, 2017-08-03:
Thanks for the suggestion, I will validate without the PTO by comparing ode14x and the default solver and respond to this thread with the results.

kmruehl, 2017-08-03:
@bradling thanks, I am interested in hearing about your solver findings. FYI, we are in the process of changing the default solver from `ode4 `to `ode45`. Have you considered running a variable time-step solver? Especially when running WEC-Sim with a coupled PTO (or mooring) model, I've found the variable solver to be more robust.

bradling, 2017-08-03:
I have looked into a variable time step solver. For my PTO only simulations I typically start with a variable time step. But this model will eventually be used in hardware-in-the-loop simulations, which requires a fixed time solver for real-time implementation.

kmruehl, 2017-08-04:
Got it, that makes sense.

kmruehl, 2017-08-18:
@bradling we finished looking into the solver options, and we can confirm that both ode4, ode45 and ode14x produce the same results within reasonable error. We performed the comparison using the application cases with the verification cases.

closed 2017-08-18 by kmruehl