WEC-Sim/WEC-Sim #192: Putting assymetric gains on excitation force?

drschandel, 2017-07-24:
Hi

I'm trying to model a highly assymetric device using WEC-Sim, which has different excitation forces acting on it in the 'forward' and 'backward' part of the waves. NEMOH can't really handle this and just outputs a linearised average value between the two. 

However I was hoping I could solve this simply by putting a 'gain' on the excitation force during the forward part of the stroke, and then tweaking the number until the device model matches experimental data. 

Is there a recommended way of doing this? Otherwise I will have to try and modify the hydrodynamic object library block with a custom gain.

Thanks
Dhruv

bjrose86, 2017-07-24:
One way might be to have WEC-Sim calculate the weakly-nonlinear excitation
forces through direct pressure integration, rather than using the
Nemoh-derived excitation forces. This can be done in WEC-Sim by setting:

simu.nlHydro = 2

Under the assumption of linear potential flow, there should not be
'forward' to 'backward' asymmetry. What you might be seeing experimentally
is a second-order wave drift force, which Nemoh or other BEM's are able to
calculate. I believe the method above should also resolve the mean drift
component.

drschandel, 2017-07-25:
Right, thanks very much, I’ll give that a try

kmruehl, 2017-07-31:
@drschandel how did running non-linear hydro work? If you wanted to modify the code to define your own non-linear forcing - through a multiplier or otherwise - you can do so. It would require modifying the bodyClass and breaking the Simulink library link. 

@bjrose86, thanks for your help!

drschandel, 2017-08-01:
Dear both, thanks for your comments. 

I didn't find non-linear hydro appropriate for this because the NEMOH coefficients themselves weren't capturing the device's physical parameters properly. So I did what @kmruehl has suggested and went into the bodyClass and modified it. It seems to be a closer match to experimental data now.

I think my problem now is actually getting an accurate PTO model rather than the hydrodynamics, since our lab PTO was very friction-ey

Thanks again for your help

closed 2017-08-01 by drschandel

kmruehl, 2017-08-01:
glad this helped.