WEC-Sim/WEC-Sim #193: Frictional resistance.

Krishnendu-Ocean, 2017-07-26:
Hello Team,

Is it possible to give a frictional force (say 3N) to a floating body restricted to heave motion alone? If so please let me know where I can provide the same.

Also, I had a query on the result obtained using WEC-Sim. I was trying to get the response of a floating body restricted to heave alone and the results showed that the floating body response profile is following the input  wave condition, that is when plotted the heave response profile of the body obtained and the wave elevation profiles are coinciding. Please note that there is no external force applied to the floating body other than the incident waves. Is this an expected response? or am I doing anything wrong with the simulations.

Awaiting for a response. 

Thanks in Advance.

Best Regards,
Krishnendu

kmruehl, 2017-07-31:
@Krishnendu-Ocean there are a few different ways to implement friction in heave - and the best way depends on your model setup/goal. What is the WEC you're modeling? Do you want to add a constant 3N force, or do want it to be dependent on the body's position/velocity? Do you want it applied directly to the body, or to the joint? 

As for the heave response, if the only force acting on the body is the wave excitation force, it makes sense that the body would be a wave follower.

Krishnendu-Ocean, 2017-08-02:
@kmruehl Thank you for your quick response.

I am trying to model a simple plate/buoy floating body restricted to heave motion as a starting step to explore WEC-Sim. I would like to know how a constant and a varying friction force can be incorporated to the body position. As far now, I have a frictional value of 3N which is constant and is applied directly to the body.

Thank you again.

kmruehl, 2017-08-02:
@Krishnendu-Ocean I recommend using a `Translational PTO Actuation Force` block between the float and spar, and adding a 3N force feeding into it. If you have to add a damping force on the float, dependent on the flaot's velocity, that can be done by defining `body(1).linearDamping`. Or if you want to add a spring or damper to the PTO dependent on the relative motion between the float and spar that can be done by defining `pto(1).k` and `pto(1).c` respectively.

closed 2017-08-02 by kmruehl

Krishnendu-Ocean, 2017-08-02:
@kmruehl  Hi, In my model there is no spar included, so should I use the Translational PTO actuation force between the translational constraint and rigid body?

kmruehl, 2017-08-02:
Sorry, I meant between the buoy and the plate. The `Translational PTO Actuation Force` should be used in place of the `Translational Constraint`

Krishnendu-Ocean, 2017-08-03:
Thanks a lot @kmruehl .