WEC-Sim/WEC-Sim #194: Question about WEC-Sim Simulink models

Wavergy, 2017-07-29:
Hi,

I'm just getting underway in trying to run a WEC-Sim simulation. I seemed to notice that all example simulations I've seen so far (such as FOSWEC, and included) tutorials only need/utilize a 2-dimensional setup when assembling the model from the WEC-Sim Library in the Simulink model.  Our device is modular-based, with adjacent surface modules serving as reaction members to each other. Is this type of 3-dimensional construction of a Simulink model possible in WEC-Sim, such that one module is modeled, in effect, behind another module with requisite joints etc being able to be constructed with the WEC-Sim library in this way.  ie: Imagine an array of floating modules on the water surface that are interconnected to eachother across the surface, and in addition interconnected to a lower

Wavergy, 2017-07-29:
Oops: cont'd

-That are also interconnected to a lower lower.

For a visual reference, please see our website:

http://Wavergy.net

Rather than a large array we would like to simulate either a 9 module array (5 on surface, 4 submerged), or a 15-module array (8 on surface, 7 submerged).

Thanks

kmruehl, 2017-07-31:
@wavergy, you can model devices in 3D in WEC-Sim just like you would model them in [Simscape Multibody](https://www.mathworks.com/products/simmechanics.html). You'll just have to define the location of each of the bodies/joints in the `wecSimInputFile.m` in <x,y,z> coordinates.

closed 2017-07-31 by kmruehl

Wavergy, 2017-07-31:
Thank you, Kelly