WEC-Sim/WEC-Sim #197: Non-existent field 'time'

jcanning11, 2017-08-28:
Hello,

I'm using the RM3 model with MoorDyn and I receive the following error after the simulation is finished:

**"Reference to non-existent field 'time'.
Error in responseClass (line 78)
obj.mooring(ii).time = mooringOutput(ii).time;"**

I believe this has something to do with choosing to not output the individual Line.out files from MoorDyn (which can be done in the lines.txt file). Is there a way around this, or will I have to include the line outputs?

Thanks,

Jarred

kmruehl, 2017-08-28:
@jcanning11 I just made some modifications to the waveClass which should resolve your previous issue. The default `waves.freqDisc` is `'Traditional'` (you can also run `'EqualEnergy'` for faster runs), and it should now work for all `'spectrumType'`. 

As for this issue, I ran the RM3 model without outputting Line1.out, Line2,out, etc and was unable to reproduce this issue. Can you try pulling the latest WEC-Sim (that I just pushed) and see if that resolves the issue. If it doesn't, can you zip your model files and upload them here?

Thank you!

kmruehl, 2017-08-29:
Issue resolved, MoorDyn runs with WEC-Sim without Line#.out files.

closed 2017-08-29 by kmruehl