WEC-Sim/WEC-Sim #199: Surge Drift

GearoidOM, 2017-08-30:
Hello,

I was reviewing the results from the comparison of WEC-Sim and Orcaflex as per results in issue #196. I was wondering why there is surge drift for the WEC-Sim model?
 
[image]

Regards,

Gearoid

yuyihsiang, 2017-08-30:
Hi Gearoid

The drift in surge is a numerical thing and probably also depends on how the ramp function and how multi-DOF coupling was implemented numerically. Basically, the model has no hydrodynamic restoring stiffness and we did not have mooring. But in WEC-Sim, there will be a very small numerically induced imbalance force in the surge direction, which will induce a small drift in surge. Since there is no hydrodynamic restoring or mooring stiffness, the model will drift. OrcaFlex probably handle the simulation differently, in terms of how the time marching method was implemented and how the "transnational joint" was applied.

Hope this answers your question,

Thanks,

--Yi-Hsiang

GearoidOM, 2017-08-31:
Thank you Yi-Hsiang

closed 2017-08-31 by GearoidOM